chipsalliance/verible #585: [lint] Flag uses of disable with anything other than `fork`

imphil, 2020-11-10:
**Summary**

Flag all uses of `disable` where it is different from `disable fork` as error.

Cadence recommends that disabling forks should be done using a `disable fork;` statement, not the `disable fork_process_label;` construct. The `disable fork_process_label` construct for forks is not defined in the IEEE Std 1800-2017 LRM. See section 9.6.2 of the LRM (2017).

**Test cases and examples**

The more examples you provide, the easier it will be for someone to come along
and implement it.

```systemverilog
// This example should be diagnosed.
disable fork_process_label;
```

```systemverilog
// This variant should NOT be diagnosed (negative test case)
disable fork;
```

**Additional context**

VCS (and likely other simulators) support `disable fork_process_label`, however, Xcelium does not. That's why this construct can go unnoticed if only VCS is used.

See also https://github.com/lowRISC/ibex/pull/1180 for an analysis of the problem, and https://github.com/lowRISC/ibex/pull/1180 for a PR fixing the cases where we had that in Ibex.

CC @PaulOKCrevinn @CathalMCrevinn did I get this description right? Please jump in with comments if needed.

fangism, 2020-11-10:
Very doable.  The CST node in question here is a `kDisableStatement`, which is constructed [here](https://cs.opensource.google/verible/verible/+/master:verilog/parser/verilog.y;drc=84495fd0f7cd301c8d71f4beeea46a1432d3d79c;l=7024).  One just has to examine those nodes' `child[1]`.

@imphil Have a passage in your style guide that can be cited (in the diagnostic message)?

Suggest new rule class named `DisableForkNoLabelsRule`.

sriyerg, 2020-11-10:
Sorry, to clarify `disable some_label` is a valid SV construct. Its just that `some_label` should not be the label of a forked thread such as `fork begin: some_label ... end join_any disable some_label;`. That may not work as expected if the thread executing the `some_label` block forks off more child threads.

sriyerg, 2020-11-10:
> Sorry, to clarify `disable some_label` is a valid SV construct. Its just that `some_label` should not be the label of a forked thread such as `fork begin: some_label ... end join_any disable [...]

Actually, its the `fork: some_label` .... `disable some_label;` that needs to be flagged. This is invalid according to the LRM.

hzeller, 2021-01-29:
So to clarify, a label that is applied to a _fork_ must not be used later in `disable`, but regular use of `disable` on some other label even in a block nested in the fork would be allowed.

```SystemVerilog
//...
fork : foo
//...
disable foo;  // Not valid, because label applied to fork
```

would not be valid. But

```SystemVerilog
fork : foo
    begin : baz
    //
        disable baz; // valid use, this is a label for our current block
    // ...
    end : baz
//...
```

Correct ?

sriyerg, 2021-01-29:
> So to clarify, a label that is applied to a _fork_ must not be used later in `disable`, but regular use of `disable` on some other label even in a block nested in the fork would be allowed. [...]

Yes that is correct.

A label applied to any sequential code block _regardless of where they appear_ can be disabled. All constructs that use `begin..end`, such as if, for, while, do, repeat, forever etc. are sequential code blocks. They all can be disabled with the `disable <label>;` statement. This is equivalent of executing the `break` where you immediately exit the code block by skipping the rest of the statements in that code block. This is perfectly legal construct and is described in section 9.6.2 of the [1800-2017 LRM](https://ieeexplore.ieee.org/document/8299595). 

`fork..join*` is not a sequential block - its a parallel block. See section 9.3 for more details. The valid way to disable all processes associated with a parallel block is using the  `disable fork` statement. The is covered in section 9.6.3. 

Both of these are legal constructs. It is upto the user to use these constructs correctly. 

What is illegal is `fork: label ... join* ... disable label;`. While it is legal to label the `fork` parallel block, it is not valid to attempt to disable it by specifying its label. 

Also, I would assume that attempting to disable process execution threads such as `initial`, `always*` and `final` using their labels:
```systemverilog
  initial begin : test
    ...
    if (cond) disable test;
  end
```
is illegal because these are not considered sequential code-blocks. The LRM makes no mention of validity of this (at least I couldn't find it) . So these will need to be flagged as well.

PR #648 (2021-05-18): disable fork_label rule for linter
Fixes  #585
closed 2021-05-18 by hzeller