chipsalliance/verible #586: [CI] migrate source indexing job to github-action

fangism, 2020-11-10:
This [kythe.sh script](https://cs.opensource.google/verible/verible/+/master:.github/kokoro/kythe.sh) is run after every commit to generate source code indexing data (uploaded to a cloud storage bucket).  Could/should we migrate this as well?  @mithro @kgugala 

This job is currently failing at top-of-tree due to a system library version mismatch on `GLIBCXX_`.  Suggested remedy is to execute those with a Docker image with the correct versions.

Details:

This release https://github.com/kythe/kythe/releases/download/v0.0.48/kythe-v0.0.48.tar.gz requires:

```
external/kythe_release/extractors/bazel_cxx_extractor: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by external/kythe_release/extractors/bazel_cxx_extractor)
external/kythe_release/extractors/bazel_cxx_extractor: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `CXXABI_1.3.11' not found (required by external/kythe_release/extractors/bazel_cxx_extractor)
external/kythe_release/extractors/bazel_cxx_extractor: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by external/kythe_release/extractors/bazel_cxx_extractor)
```

fangism, 2020-11-19:
With #608 in, is it now possible to use one of our generated Docker images to kythe-index C++ sources, instead of whatever is currently on the kokoro executor?

fangism, 2020-12-10:
This will be fixed by #631

mithro, 2020-12-10:
There is still a little more work to make the GitHub action upload the kzips to somewhere that codesearch and fetch them from (GCP).

commit 3363b2c9 (2020-12-10): PR #631: kythe: Adding GitHub Action for Kythe run.

GitHub PR https://github.com/google/verible/pull/631

Copybara import of the project:

  - 1b682f0ca1b4d7f0dcf6b56d3734494d91c698bf gha: Small change to yaml file formatting. by Tim 'mithro' Ansell <me@mith.ro>
  - a51ea1ceda193ed1448f2cf16ec6f4137a3b0a65 kythe: Adding GitHub Action for Kythe run. by Tim 'mithro' Ansell <me@mith.ro>
  - 5bd74f04b6500579895866c71c566df7e6e9374c kythe: Trying ubuntu-20.04 by Tim 'mithro' Ansell <me@mith.ro>
  - aac359ec274292cc92e0bb72383204a0e9ed17c6 gha: Give the uploaded artifacts names. by Tim 'mithro' Ans [... cut]