chipsalliance/verible #595: [kythe] linkify macro definitions' use of macro formal arguments

fangism, 2020-11-10:
## Input test case

Try to reduce the test down to a single demonstrative file.

```systemverilog
`define uvm_create_comp(_type_name_, _inst_name_) \
  _inst_name_ = _type_name_::type_id::create(`"_inst_name_`", this);
```

## Describe what is wrong or missing

Formal arguments are `_type_name_` and `_inst_name_`.
Inside the definition body, **only** linkify those, because other names cannot be resolved without macro call-site context.

You will need to [recursively lex](https://cs.opensource.google/verible/verible/+/master:verilog/parser/verilog_lexer.h;bpv=1;bpt=1;l=73?q=RecursiveLex&ss=verible%2Fverible&gsn=RecursiveLexText&gs=kythe%3A%2F%2Fgithub.com%2Fgoogle%2Fverible%3Flang%3Dc%252B%252B%3Fpath%3Dverilog%2Fparser%2Fverilog_lexer.h%23C0nKQmC4ZWIOKHhkIfITTvWJi-cWnRTde95MvNDv1bI) the macro definition body.

MinaToma, 2020-11-11:
Maybe a duplicate of this one: #425
@fangism feel free to close one of them.

fangism, 2020-11-11:
> Maybe a duplicate of this one: #425
> @fangism feel free to close one of them.

This one is slightly different, it is only about formal arguments.

fangism, 2020-11-11:
Note that references to macro formal arguments look like `SymbolIdentifier` and `EvalStringLiteral` when recursively lexing inside macro definition bodies.