chipsalliance/verible #605: always-ff-non-blocking rule wrongly flags assignments to locals

preusser, 2020-11-16:
**Summary & Example**

With the exception of loop headers, the [always-ff-non-blocking](https://github.com/google/verible/blob/73936a870a7948663cc1009cd8b36a40f3ebe69d/verilog/analysis/checkers/always_ff_non_blocking_rule.cc#L56) currently flags all blocking assignments in an `always_ff` block regardless of the nature of the targeted variable. This disallows structuring non-trivial computations into steps using `automatic` locals, such as:
```systemverilog
always_ff (@posege clk) begin
  automatic type(Count)  count_nxt = Count;
  if(inc_valid)  count_nxt = count_nxt + inc_value;      // Should not be flagged
  if(dec_valid)  count_nxt = count_nxt - dec_value;      // Should not be flagged
  Count <= count_nxt;                                    // Would be flagged if blocking
  Trigger <= count_nxt >= THRESHOLD;                     // Would be flagged if blocking
end
```

**Proposal**
The rule should be waived for assignments to locally declared variables, which cannot be observed by unrelated processes.

The linter can do so when knowing the variable declarations of local, possibly nested blocks of the containing `always_ff` statement. Code tracking the declarations of variables in these local scopes and waiving the rule accordingly would be available for review.

fangism, 2020-11-16:
Good point!  @snsokolov What do you think?

snsokolov, 2020-11-16:
> Good point! @snsokolov What do you think?

The style of locally declared automatic variables inside always_ff is rarely used. I think we recommend explicitly declaring these as nets and combinational logic outside of always_ff blocks. Let me ask this internally.

preusser, 2020-11-16:
This would be a recommendation against tight scoping. And even if so, the proposed modification would not invalidate such a style.

snsokolov, 2020-11-17:
Hi Thomas,

I checked internally and confirmed that for various reasons (risk of inferred latch is one of them I think) our internal style guide explicitly prohibit mixing any (especially non-trivial) combo logic with sequential logic within the same a sequential block. And I don't think we would change this guidance.

By the way, what about waveforms? Is this okay from debuggability standpoint that `count_nxt` is not visible to waveforms?

Off to David.

fangism, 2020-11-17:
While we may not be in a good position to change the existing rule, we're not opposed to anyone making a different version of the rule that grants exceptions like the proposed to fit your own style guide.  (Someday making them configurable variants of the same rule.). That being said, the exception requested requires technically:

* More bookkeeping for what constitutes local variables to represent convenient subexpressions (as you said).  This could range from having a full symbolic resolution, or a more impromptu tracking of local declarations.
* Deeper examination of right-hand-side expressions.  Should an expression mixed with local/state-holding variables be diagnosed?

Style-wise: Does anyone ever push non-trivial logic into functions?  That would give the benefit of tight-scoping around local variables, and still be (my opinion) readable.

```systemverilog
  count <= func(count, state, inputs);
```

preusser, 2020-11-17:
@snsokolov:
- As `automatic` variables do not retain their state between block invocations, they *cannot* imply latches.
- I would even claim that latches are not created from improper sequential but rather from improper combinatorial logic.
- Yes, observability is reduced for local variables as you are moving from a strictly structural to a more behavioral description. We do not advise to bury complex computations this way. On the other hand, there is a wide range between trivial and complex. So, we encourage thoughtful trade-offs allowing higher-level behavioral descriptions where appropriate.

@fangism:
- I am happy to propose the modified semantics as a new independent rule. As I said, the code is there, and I would welcome your feedback.
- I do not see any good reason to add constraints to the composition of the right-hand side. In the end, `automatic` locals are just extracted and named subexpressions.
- Functions are interesting for re-using behavior and for wrapping semantically coherent behavior under a headline. However, if used as an instrument to circumvent coding guidelines, re-use will be low and crisp headlines will be rare. Nonetheless, you will be paying the price of fragmenting the behavioral description by externalizing very few lines at a time. While the linter might be happy, such code is typically rather hard to read.

fangism, 2020-11-17:
Sure, if you have code for a new rule already, we can review it, send a pull-request.  For style lint rules, we usually encourage a discussion around proposed test cases first.

commit d91a8d53 (2021-02-17): Add config to always_ff_non_blocking: allow local assignments, catch_modifying_assignments (#634)

Add configuration flags to always_ff_non_blocking
 * `waive_for_locals` to not flag local assignments (fixes #605 )
 * `catch_modifying_assignments`: Provide a stricter test to catch modifying assignments (fixes #606) 
  
Fixes: #605 and #606:
closed 2021-02-17 by hzeller