chipsalliance/verible #611: /usr/include/c++/5/bits/stl_pair.h:108:26: note: candidate expects 0 arguments, 2 provided

mithro, 2020-11-18:
The following error is causing Verbile to not build on Ubuntu Xenial;
```
/usr/include/c++/5/ext/new_allocator.h:120:4: error: no matching function for call to 'std::pair<const std::__cxx11::basic_string<char>, verible::SubcommandEntry>::pair(absl::lts_2020_09_23::string_view&, const verible::SubcommandEntry&)'
  { ::new((void *)__p) _Up(std::forward<_Args>(__args)...); }
    ^
In file included from /usr/include/c++/5/utility:70:0,
                 from /usr/include/c++/5/tuple:38,
                 from /usr/include/c++/5/functional:55,
                 from ./common/util/subcommand.h:18,
                 from common/util/subcommand.cc:15:
/usr/include/c++/5/bits/stl_pair.h:206:9: note: candidate: template<class ... _Args1, long unsigned int ..._Indexes1, class ... _Args2, long unsigned int ..._Indexes2> std::pair<_T1, _T2>::pair(std::tuple<_Args1 ...>&, std::tuple<_Args2 ...>&, std::_Index_tuple<_Indexes1 ...>, std::_Index_tuple<_Indexes2 ...>)
         pair(tuple<_Args1...>&, tuple<_Args2...>&,
         ^
/usr/include/c++/5/bits/stl_pair.h:206:9: note:   template argument deduction/substitution failed:
In file included from /usr/include/x86_64-linux-gnu/c++/5/bits/c++allocator.h:33:0,
                 from /usr/include/c++/5/bits/allocator.h:46,
                 from /usr/include/c++/5/string:41,
                 from /usr/include/c++/5/stdexcept:39,
                 from /usr/include/c++/5/array:38,
                 from /usr/include/c++/5/tuple:39,
                 from /usr/include/c++/5/functional:55,
                 from ./common/util/subcommand.h:18,
                 from common/util/subcommand.cc:15:
/usr/include/c++/5/ext/new_allocator.h:120:4: note:   'absl::lts_2020_09_23::string_view' is not derived from 'std::tuple<_Elements ...>'
  { ::new((void *)__p) _Up(std::forward<_Args>(__args)...); }
    ^
In file included from /usr/include/c++/5/utility:70:0,
                 from /usr/include/c++/5/tuple:38,
                 from /usr/include/c++/5/functional:55,
                 from ./common/util/subcommand.h:18,
                 from common/util/subcommand.cc:15:
/usr/include/c++/5/bits/stl_pair.h:155:9: note: candidate: template<class ... _Args1, class ... _Args2> std::pair<_T1, _T2>::pair(std::piecewise_construct_t, std::tuple<_Args1 ...>, std::tuple<_Args2 ...>)
         pair(piecewise_construct_t, tuple<_Args1...>, tuple<_Args2...>);
         ^
/usr/include/c++/5/bits/stl_pair.h:155:9: note:   template argument deduction/substitution failed:
In file included from /usr/include/x86_64-linux-gnu/c++/5/bits/c++allocator.h:33:0,
                 from /usr/include/c++/5/bits/allocator.h:46,
                 from /usr/include/c++/5/string:41,
                 from /usr/include/c++/5/stdexcept:39,
                 from /usr/include/c++/5/array:38,
                 from /usr/include/c++/5/tuple:39,
                 from /usr/include/c++/5/functional:55,
                 from ./common/util/subcommand.h:18,
                 from common/util/subcommand.cc:15:
/usr/include/c++/5/ext/new_allocator.h:120:4: note:   cannot convert 'std::forward<absl::lts_2020_09_23::string_view&>((* & __args#0))' (type 'absl::lts_2020_09_23::string_view') to type 'std::piecewise_construct_t'
  { ::new((void *)__p) _Up(std::forward<_Args>(__args)...); }
    ^
In file included from /usr/include/c++/5/utility:70:0,
                 from /usr/include/c++/5/tuple:38,
                 from /usr/include/c++/5/functional:55,
                 from ./common/util/subcommand.h:18,
                 from common/util/subcommand.cc:15:
/usr/include/c++/5/bits/stl_pair.h:150:12: note: candidate: template<class _U1, class _U2, class> constexpr std::pair<_T1, _T2>::pair(std::pair<_U1, _U2>&&)
  constexpr pair(pair<_U1, _U2>&& __p)
            ^
/usr/include/c++/5/bits/stl_pair.h:150:12: note:   template argument deduction/substitution failed:
In file included from /usr/include/x86_64-linux-gnu/c++/5/bits/c++allocator.h:33:0,
                 from /usr/include/c++/5/bits/allocator.h:46,
                 from /usr/include/c++/5/string:41,
                 from /usr/include/c++/5/stdexcept:39,
                 from /usr/include/c++/5/array:38,
                 from /usr/include/c++/5/tuple:39,
                 from /usr/include/c++/5/functional:55,
                 from ./common/util/subcommand.h:18,
                 from common/util/subcommand.cc:15:
/usr/include/c++/5/ext/new_allocator.h:120:4: note:   'absl::lts_2020_09_23::string_view' is not derived from 'std::pair<_T1, _T2>'
  { ::new((void *)__p) _Up(std::forward<_Args>(__args)...); }
    ^
In file included from /usr/include/c++/5/utility:70:0,
                 from /usr/include/c++/5/tuple:38,
[... 58 lines cut ...]
                 from common/util/subcommand.cc:15:
/usr/include/c++/5/ext/new_allocator.h:120:4: note:   'absl::lts_2020_09_23::string_view' is not derived from 'const std::pair<_T1, _T2>'
  { ::new((void *)__p) _Up(std::forward<_Args>(__args)...); }
    ^
In file included from /usr/include/c++/5/utility:70:0,
                 from /usr/include/c++/5/tuple:38,
                 from /usr/include/c++/5/functional:55,
                 from ./common/util/subcommand.h:18,
                 from common/util/subcommand.cc:15:
/usr/include/c++/5/bits/stl_pair.h:112:26: note: candidate: constexpr std::pair<_T1, _T2>::pair(const _T1&, const _T2&) [with _T1 = const std::__cxx11::basic_string<char>; _T2 = verible::SubcommandEntry]
       _GLIBCXX_CONSTEXPR pair(const _T1& __a, const _T2& __b)
                          ^
/usr/include/c++/5/bits/stl_pair.h:112:26: note:   no known conversion for argument 1 from 'absl::lts_2020_09_23::string_view' to 'const std::__cxx11::basic_string<char>&'
/usr/include/c++/5/bits/stl_pair.h:108:26: note: candidate: constexpr std::pair<_T1, _T2>::pair() [with _T1 = const std::__cxx11::basic_string<char>; _T2 = verible::SubcommandEntry]
       _GLIBCXX_CONSTEXPR pair()
                          ^
/usr/include/c++/5/bits/stl_pair.h:108:26: note:   candidate expects 0 arguments, 2 provided
INFO: Elapsed time: 608.097s, Critical Path: 132.53s
INFO: 1240 processes: 391 internal, 849 processwrapper-sandbox.
FAILED: Build did NOT complete successfully
FAILED: Build did NOT complete successfully
The command '/bin/sh -c bazel build --workspace_status_command=bazel/build-version.sh -c opt --show_progress_rate_limit=10.0 //...' returned a non-zero code: 1

Error: Process completed with exit code 1.
```

mithro, 2020-11-18:
@hzeller -- Any idea?

wtatarski, 2020-11-18:
@mithro  try to install gcc-7 and g++-7. It worked on my PR.

fangism, 2020-11-18:
This line numbers are a little off, but I think this is coming from the `emplace` call here:

https://cs.opensource.google/verible/verible/+/master:common/util/subcommand.cc;bpv=1;bpt=1;l=64?gsn=RegisterCommand&gs=kythe%3A%2F%2Fgithub.com%2Fgoogle%2Fverible%3Flang%3Dc%252B%252B%3Fpath%3Dcommon%2Futil%2Fsubcommand.cc%23YYdzeZJvf-rXeM3FY64I6pXTp8ReRf4sD3UprqjccAI&gs=kythe%3A%2F%2Fgithub.com%2Fgoogle%2Fverible%3Flang%3Dc%252B%252B%3Fpath%3Dcommon%2Futil%2Fsubcommand.h%23LUSjZoR2HrCV9QFNv1lPdumicX-uKjL51XbPfsbZAkU

C++11 didn't have heterogenous lookup until C++14, but the `#if __cplusplus >= 201402L` was intended to handle that, and fall back to `std::string` construction, but maybe that constructor is missing in an old stdlib version.

Possible fix: `std::string(name.begin(), name.end())` (2-iterator range constructor)

commit af4ac52e (2020-11-20): Support older compiler/library use of std::string/string_view.

Older libraries don't support direct conversion from absl::string_view to
std::string.  Use 2-iterator range construction instead.

fixes #611

PiperOrigin-RevId: 343297893
closed 2020-11-20 by fangism