chipsalliance/verible #617: Data type check in function arguments

svenka3, 2020-12-02:
**Summary**
Recently found this on OpenTitan, see:

https://github.com/lowRISC/opentitan/issues/4355#issuecomment-736782490

Wondering if this will be caught in Verible using existing rules. 

**Test cases and examples**

Is there a running version of OpenTitan in Verible that I can try out at my end? 

Code snippet is available on https://github.com/lowRISC/opentitan/issues/4355#issuecomment-736782490

Will be happy to extract a small test should that be necessary.

fangism, 2020-12-02:
Verible is a still quite a ways off from being able to do semantics that require elaboration.  Type/width checking will have to come after symbol-binding (we have a small start in the way of source code indexing), and preprocessing.  I believe Verilator comes with many semantic linting capabilities.

svenka3, 2020-12-02:
OK, will close this ticket.

closed 2020-12-02 by svenka3

mithro, 2020-12-02:
https://github.com/alainmarcel/Surelog might be able to do what you want.