chipsalliance/verible #618: Verible - policy/rule-set for UVM migration

svenka3, 2020-12-02:
**Summary**

Based on an ongoing porting of OpenTitan across simulators, we are finding some interesting issues on compatibility. Wondering if we can extract a rule-set/policy and make it part of Verible add-ons to say:

SIM-1 to SIM-2 migration style check 

Any thoughts on how this can be approached in Verible context please?

fangism, 2020-12-02:
Today, Verible is capable of doing syntax-based checking, but is missing semantics -- if you find a particular language feature risky enough to ban, and it is detectable as a matter of syntax, Verible is up to the task.  As other SIM-tools improve over time, the compatibility matrix may also change.

I would however recommend getting involved with https://symbiflow.github.io/sv-tests/ to contribute to the suite of test cases that are checked against a variety of tools.  It's an ever growing public suite of tests that covers syntax, and gradually, more semantics.  While users may not be at liberty to publish findings about differences between commercial tools, it is certainly available for users to test for themselves.  And also bear in mind that not everything that is different is wrong, the SV standard leaves room for interpretation and implementation-defined choices.

svenka3, 2020-12-02:
Thanks, yes am starting to add tests that wonderful sv-tests repo.

Cheers,

closed 2020-12-02 by svenka3

mithro, 2020-12-02:
Just FYI - https://github.com/alainmarcel/Surelog might be able to do more complicated checks (and Google is also helping fund the project). It however is significantly slower and more complex than verible.