chipsalliance/verible #619: Releases seem to be broken

mithro, 2020-12-04:
It seems like the uploading github releases is not working.

fangism, 2020-12-04:
Is this related to the kokoro-kythe faillures?

mithro, 2020-12-04:
I think is caused by the Travis -> GitHub actions move...

mithro, 2020-12-04:
All the released files seem to be uploading to the latest release -- https://github.com/google/verible/releases

mithro, 2020-12-04:
[image]

mithro, 2020-12-04:
* https://github.com/marketplace/actions/create-a-release
 * https://github.com/marketplace/actions/upload-a-release-asset

mithro, 2020-12-04:
We should probably be using one of the two more standard github actions for uploading a the releases. I'm unsure why we are using https://github.com/svenstaro/upload-release-action

https://github.com/google/verible/blob/b74ff0df4ae9e9d0457cefaa3e6774f4b1251d6f/.github/workflows/verible-ci.yml#L85-L93

mithro, 2020-12-05:
The issue is https://github.com/google/verible/blob/b74ff0df4ae9e9d0457cefaa3e6774f4b1251d6f/.github/workflows/verible-ci.yml#L91

`${{ github.ref }}` converts to `master` rather than the newly generated tag version.

umarcor, 2020-12-05:
Is the purpose the following?

On sucess:

- Create a git tag.
- Push the tag to the repo.
- Create a release pointing to that tag.
- Upload some assets to the release.

mithro, 2020-12-05:
Pretty much.

What I don't understand is how the `$TRAVIS_TAG` value ended up being `# Create a tag of form v0.0-183-gdf2b162-20191112132344`

mithro, 2020-12-05:
`git-describe` would produce the `v0.0-183-gdf2b162` part and the number comes from when the travis job was scheduled...

mithro, 2020-12-05:
Looking at all the releases listed, as far as I can tell the date part has never worked.....

closed 2020-12-08 by hzeller

mithro, 2020-12-08:
Looks like after #625 was fixed, this change started working! https://github.com/google/verible/releases
[image]

umarcor, 2020-12-08:
@mithro, remember to remove tag 'master' and that release with 142 assets.

commit acb8a1a1 (2020-12-08): PR #620: Fixing release publishing
Fixes #619.

GitHub PR https://github.com/google/verible/pull/620

Copybara import of the project:

  - 5f9b06d1c0a30e4c662b5bce7d4b51c2afe982a2 Revert incorrect change no longer using `git describe`. by Tim 'mithro' Ansell <me@mith.ro>
  - 8aaeca17158b0d8f13b77a4667ee51214ff5edb3 gha: Export and then use tag value. by Tim 'mithro' Ansell <me@mith.ro>
  - b5a598fca0b02b5b6bf3439ba1b9c1eaea57735c gha: Use github cache. by Tim 'mithro' Ansell <me@mith.ro>
  - f6f93ae62313058f265507926229d60908cdf09b Merge b5a598fca0b02b5b6bf3439ba1b9c1eaea57735c into b74ff... b [... cut]