chipsalliance/verible #621: Generate the list of GitHub release CI jobs automagically rather than hard coding in the workflow

mithro, 2020-12-05:
I believe it is possible to have a GitHub Action generate follow up GitHub Action steps. It would be nice if the GitHub Action for building releases just understood the currently supported releases rather than requiring the workflow to be updated.

mithro, 2020-12-05:
I believe @umarcor might know how to do this.

umarcor, 2020-12-08:
So, yes and no. It is possible to generate/execute a set of tasks and wrap them in collapsible log blocks. See, for instance, step "*Build and test GHDL in containers*" in https://github.com/ghdl/ghdl/runs/1515138857?check_suite_focus=true or the test script in fomu-workshop: https://github.com/im-tomu/fomu-workshop/runs/1509392137?check_suite_focus=true#step:10:8. However, AFAIAA it is not possible to generate jobs or steps programmatically. That's a purely descriptive thing: what you write in the YAML is what is picked by the web GUI. If that was to be done, some preprocessor/sed would be required as a commit hook.

Overall, I think that listing the release jobs explicitly is not a big deal. That allows better visualisation and it also allows limiting the number of parallel jobs. However, the CI plumbing feels awkward to me. It seems that all the dockerfiles and archives are generated through a shell scrip, in each job; even though only one of the subdirs is used. Then, each job pushes one artifact and attempts to create a release (with one asset).

tgorochowik, 2020-12-08:
You can generate the matrix programmatically and then use it to generate jobs dynamically:
See this as an example: https://github.com/alainmarcel/uhdm-integration/blob/master/.github/workflows/main.yml

One thing to note is that there is a limit of 256 jobs per matrix.

umarcor, 2020-12-08:
@tgorochowik I was not aware of that. Thanks a lot for the reference!

commit dc6e959a (2020-12-08): PR #623: ci: generate Release matrix dynamically
Close #621.

As suggested by @mithro and explained by @tgorochowik, this PR generates the Release matrix dynamically.

GitHub PR https://github.com/google/verible/pull/623

Copybara import of the project:

  - 05b9c0be17ddb0b65ffdff2d75e8f2b2b45008a0 ci: generate Release matrix dynamically by umarcor <unai.martinezcorral@ehu.eus>
  - 6b61ec994ad0856656698ae802ec68b622a11a9c releasing: disable centos-6 by umarcor <unai.martinezcorral@ehu.eus>
  - 8b16f4ccff03c6637c0de166b4b2bbc11c024354 ci: add script 'generate_matrix.sh' by umarcor <unai.martine [... cut]
closed 2020-12-08 by fangism