chipsalliance/verible #627: [CI] only Centos8 is failing tar archiving

fangism, 2020-12-08:
See https://github.com/google/verible/runs/1520501401 for example

mithro, 2020-12-08:
`2020-12-08T23:02:24.4735337Z [91m/out//verible-v0.0-800-gdc6e959-CentOS-8.3.2011-n/a-x86_64.tar.gz: Cannot open`

mithro, 2020-12-08:
Issue is the `n/a` in the output.

mithro, 2020-12-09:
Can repo locally;
```
        not a dynamic executable
Running: /out//verible-v0.0-800-gdc6e959/bin/verible-verilog-syntax --helpfull </dev/null 2>&1
Writing: /out//verible-v0.0-800-gdc6e959/share/man/man1/verible-verilog-syntax.1
Running: /out//verible-v0.0-800-gdc6e959/bin/verible-verilog-lint --helpfull </dev/null 2>&1
Writing: /out//verible-v0.0-800-gdc6e959/share/man/man1/verible-verilog-lint.1
Running: /out//verible-v0.0-800-gdc6e959/bin/verible-verilog-format --helpfull </dev/null 2>&1
Writing: /out//verible-v0.0-800-gdc6e959/share/man/man1/verible-verilog-format.1
Running: /out//verible-v0.0-800-gdc6e959/bin/verible-verilog-diff --helpfull </dev/null 2>&1
Writing: /out//verible-v0.0-800-gdc6e959/share/man/man1/verible-verilog-diff.1
Running: /out//verible-v0.0-800-gdc6e959/bin/verible-verilog-obfuscate --helpfull </dev/null 2>&1
Writing: /out//verible-v0.0-800-gdc6e959/share/man/man1/verible-verilog-obfuscate.1
tar (child): /out//verible-v0.0-800-gdc6e959-CentOS-8.3.2011-n/a-x86_64.tar.gz: Cannot open: No such file or directory
tar (child): Error is not recoverable: exiting now
verible-v0.0-800-gdc6e959/
verible-v0.0-800-gdc6e959/bin/
verible-v0.0-800-gdc6e959/bin/verible-verilog-syntax
tar: /out//verible-v0.0-800-gdc6e959-CentOS-8.3.2011-n/a-x86_64.tar.gz: Cannot write: Broken pipe
tar: Child returned status 2
tar: Error is not recoverable: exiting now
The command '/bin/sh -c ./.github/workflows/github-releases-setup.sh /out/' returned a non-zero code: 2
```

mithro, 2020-12-09:
I think the cause is `DISTRO_CODENAME=$(lsb_release --short --codename)`

mithro, 2020-12-09:
https://github.com/google/verible/blob/5c342396f15e6e0df787dc2ebc7a3986f8640706/.github/workflows/github-releases-setup.sh#L57

commit 23eebd86 (2020-12-09): PR #629: releasing: Make sure the codename is valid.
Fixes #627.

GitHub PR https://github.com/google/verible/pull/629

Copybara import of the project:

  - 2075a58a6ee76feaa8b7da7858f024f70c3685d1 releasing: Make sure the codename is valid. by Tim 'mithro' Ansell <me@mith.ro>
  - 81c30e277cc7ce91fe6df67dafc57fefd72fea38 Merge 2075a58a6ee76feaa8b7da7858f024f70c3685d1 into 5c342... by Tim Ansell <me@mith.ro>

Closes #629

PiperOrigin-RevId: 346493887
closed 2020-12-09 by fangism