chipsalliance/verible #636: Issues with google.github.io/verible/ web page

snsokolov, 2021-01-05:
1. https://google.github.io/verible/AUTHORS.html is not properly formatted. It looks like the txt file is [just copied](https://github.com/google/verible/blob/c853a75ad833883506b8205d2229ad37831e84cc/.github/workflows/github-pages-setup.sh#L36) into and md file. The file content needs to be wrapped with ``` before making it an MD file.
2. Link to [SV tests](https://symbiflow.github.io/sv-tests/) is not highlighted at https://google.github.io/verible/. It looks like github is not auto-detecting these links in web view, we may want to mark these links using []() notation explicitly
3. Link to https://cs.opensource.google/verible/verible is not highlighted at https://google.github.io/verible/README.html
4. A reference to SV-LRM is broken at https://google.github.io/verible/ because the footer of the file with the link (https://github.com/google/verible/blame/master/README.md#L179)  is missing when the index.md is generated.

snsokolov, 2021-01-05:
5. Also many broken links at https://google.github.io/verible/README.html
https://google.github.io/verible/verilog/parser
https://google.github.io/verible/verilog/tools/lint
https://google.github.io/verible/verilog/tools/diff
etc.

fangism, 2021-01-05:
In general, we could use some form of generated-documentation validation that includes link checking (best if this is possible before actually pushing a commit, and verified in a staging environment).

snsokolov, 2021-01-05:
Not committing yet, but I'll try to take a look at it this week. Unless someone else wants to fix that.

mithro, 2021-01-05:
FYI - I just threw together the quickest possible solution to get *something* going.

Long term all our system verilog related hardware projects are using Sphinx (https://www.sphinx-doc.org/en/master/) with ReadTheDocs (https://readthedocs.org/) and a bunch of custom extensions like http://sphinxcontrib-verilog-diagrams.rtfd.io/ -- a super example is the Verilog to Routing documentation at https://docs.verilogtorouting.org/ or the SymbiFlow documentation at http://symbiflow.rtfd.io/