chipsalliance/verible #639: Structure literal formatting issues

zegervdv, 2021-01-11:
**Test case**

```systemverilog
module foobar ();

always_comb begin
some_test_structure = '{
  some_long_field_name: signal_one_input + 3 * (signal_two_input + signal_three_input + 3),
  some_other_field_name: '{
    sub_field_name: signal_four_input[$clog2(foobar * 3)-1:0] + (signal_five_input << 2),
    second_sub_field_name: signal_six_input + 4
  },
  last_name: signal_seven
};
end
endmodule

```

Include any options or configuration used.

Using default options: `verible-verilog-format -inplace test.sv`

**Actual output**

```systemverilog
module foobar ();

  always_comb begin
    some_test_structure = '{
  some_long_field_name: signal_one_input + 3 * (signal_two_input + signal_three_input + 3),
  some_other_field_name: '{
    sub_field_name: signal_four_input[$clog2(
        foobar*3
    )-1:0] + (signal_five_input << 2), second_sub_field_name: signal_six_input + 4},
        last_name: signal_seven};
  end
endmodule

```

Include any possible diagnostic messages from the formatter.

**Expected or suggested output**

```systemverilog
module foobar ();

  always_comb begin
    some_test_structure = '{
      some_long_field_name: signal_one_input + 3 * (signal_two_input + signal_three_input + 3),
      some_other_field_name: '{
        sub_field_name: signal_four_input[$clog2(foobar*3)-1:0] + (signal_five_input << 2),
        second_sub_field_name: signal_six_input + 4   // Prefer splitting fields to a new line over splitting the RHS
      }, // closing bracket on new line
      last_name: signal_seven // Correct indentation
    };  // closing bracket on new line
  end
endmodule

```

I'm seeing multiple issues with structure literals, which seem to have gotten worse somewhere since version `v0.0-589-g876c18b`.
I am using the latest version `v0.0-857-gd4bd110`.

When using a structure literal with field names, the names are not indented properly on the first level, note that the first field name of the inner struct is correctly indented.
The closing `}` is also being appended to the last line in stead of being on its own line.
Fields of the outer structure are also not aligned consistently anymore after the nested structure literal is closed.

Also the breaking of the slice seems unnecessary here.

fangism, 2021-01-11:
Nice test case.  It is know that the wrapping implementation needs a lot of work.  
@ldalek-antmicro FYI

zegervdv, 2022-08-25:
I've run the test again on `v0.0-2135-gb534c1fe` and the formatter outputs cleaner code now:
``` systemverilog
module foobar ();

  always_comb begin
    some_test_structure = '{
        some_long_field_name: signal_one_input + 3 * (signal_two_input + signal_three_input + 3),
        some_other_field_name: '{
            sub_field_name: signal_four_input[$clog2(foobar*3)-1:0] + (signal_five_input << 2),
            second_sub_field_name: signal_six_input + 4
        },
        last_name: signal_seven
    };
  end
endmodule
```

I think this can be closed.