chipsalliance/verible #653: Add clang-format check in pull requests

hzeller, 2021-01-27:
Let's add another check to the presubmit checks: run all modified/new c++ files through `clang-format --style=Google` and report back if there is a diff.

Ask the user to run clang-format on the code and update the pull request in that case.

(We have to report and can probably not update the pull request in-place with clang-format, because that might mess with a WIP pull request.)

hzeller, 2021-01-27:
But maybe we could just run `clang-format` as a hook at time of merge ? That way, we don't have to bother the contributor with an insignificant technicality (we could still make this a pull request check, but not a merge-blocking one).

In the past operating model we were actually [doing this](https://github.com/google/verible/blob/b343939bb233786cca3469134c414ca40312fdfc/CONTRIBUTING.md#style), but we have to build out these things without the additional round-trip.

mithro, 2021-01-27:
@hzeller -- I believe there are github actions that can be set up which will add a patch onto people's pull requests with the reformatting (or maybe just a set of suggestions?).

mithro, 2021-01-27:
@umarcor -- Thoughts?

umarcor, 2021-01-28:
It is technically possible to have a GitHub Actions workflow/job which interacts with either git or the GitHUb API for:

- Editing the original message in the PR.
- Posting a comment in the PR, or editing an already posted comment by the bot.
- Committing to the PR or to any other repository anywhere.

However, the scope of the default token is limited, and it does not suffice for pushing commits on top of existing PRs from forks. It will only work for PRs created from feature branches in this same repo.

Updating any PR would require the Personal Access Token of some maintainer to be added as a CI secret. That is undesirable, because then the scope is too wide and it's a potential security issue.

I would recommend avoiding having PRs updated automatically through this mechanism. If code style is considered code quality, IMHO it should be a regular job that fails. Contributors should take care of running clang-format and clang-tidy (through a hook, if they want to). Alternatively, if the style is not considered critical, I would not add any CI check, and just let maintainers reformat periodically.

mithro, 2021-01-28:
BTW -- GitHub has a way to "suggesting" changes in a comment that the owner of the pull request can accept.

umarcor, 2021-01-28:
It has. However, I think that each suggested change is saved as a different commit. It makes squashing almost a requirement... I find it desirable instead for CI to fail and contributors to run formatting locally.

hzeller, 2021-01-28:
Yes, I think having a simple CI test that fails will be a simple and robust way to deal with this. It also allows the contributor to do everything on their machine with a one-liner which is much more comfortable than clicking many buttons in a browser.

hzeller, 2021-02-04:
`clang-format` is covered. Remains `clang-tidy` to run on a pull request.
For this, `clang-tidy` needs a [compilation database](https://clang.llvm.org/docs/JSONCompilationDatabase.html), essentially containing compiler invocation parameters for the files involved.

It looks like [bazel]() does not support this (yet?) natively (I think), but there is some  IDE integration https://blog.bazel.build/2016/06/10/ide-support.html bridge code. Also there is the output of `bazel build -s ...` that could probably be used.

This project looks promising: https://github.com/grailbio/bazel-compilation-database : it provides a bazel rule to generate a compilation database, though it looks like it requires to list all the cc library rules ? Anyway looks like a good starting point.

tgorochowik, 2021-02-05:
BTW Verilator does this, it creates an extra commit:
https://github.com/verilator/verilator/blob/36eb952b60239e0f42a055622b8273cabc6e1c84/.github/workflows/clang-format.yml
I think the permission problem is solved if such commits are pushed after the PR is merged (so creating PRs from different organizations is not a problem)

This is probably not the prettiest solution due to a bunch of reformatting commits in the master branch (e.g. https://github.com/verilator/verilator/commit/7bce15b065c2c305f39168c2b9ae0d28fa5ee801), but it seems to be a low-maintenance solution that just works.

PR #914 (2021-09-07): Add clang-tidy script, to be run from the CI.
  * Use bazel rule to generate compilation database.
  * Run clang-tidy with some initial useful warnings enabled.
    The output file is currently just displayed in the CI log,
    but once that is clean, we can fail-on-warning

Fixes: #653
commit 8a55266c (2021-09-07): Add clang-tidy script, to be run from the CI.

  * Use bazel rule to generate compilation database.
  * Run clang-tidy with some initial useful warnings enabled.
    The output file is currently just displayed in the CI log,
    but once that is clean, we can fail-on-warning

Fixes: #653
closed 2021-09-07 by hzeller