chipsalliance/verible #657: Code under else if is folded into comments.

cxcxcxcx, 2021-01-27:
**Test case**

```systemverilog
// Input to the formatter, preferably a reduced test case.
module abcd;                                 
 always @(posedge clk or negedge rst_n) begin
   if (~rst_n) begin                         
     d <= 1;                                 
   end else if (abc)                         
     // comment                              
     d <= 4;                                 
 end                                         
endmodule                                    
```

Include any options or configuration used.

**Actual output**

```
-: Error lex/parsing-ing formatted output.  Please file a bug.      
First error: token: "end" at 6:3-5; problematic formatter output is 
module abcd;                                                        
  always @(posedge clk or negedge rst_n) begin                      
    if (~rst_n) begin                                               
      d <= 1;                                                       
    end else if (abc)  // comment     d <= 4;                       
  end                                                               
endmodule                                                           
<<EOF>>                                                             
```

hzeller, 2021-01-29:
Good find. Luckily the formatter will make sure to fail-safe and _not_ modify the file when it discovered the discrepancy.

We can reduce that even further; it is not necessarily the 'else' part, but just the fact that a single line 'block' is spanning multiple lines of which there is an end-of line comment is then folded into a single line:

```SystemVerilog
module abcd;                                 
 always @(posedge clk or negedge rst_n) begin
   if (~rst_n)                         
     // comment in multiline without begin/end
     d <= 1;                                 
 end                                         
endmodule        
```

(the diagnostic output then shows that it would arrive at the following and thus skips the formatting for this file)

```SystemVerilog
module abcd;
  always @(posedge clk or negedge rst_n) begin
    if (~rst_n)  // comment in multiline without begin/end     d <= 1;
  end
endmodule
```

Humans will also have a problem reading such code, so adding a begin/end around such code that needs more than one line is probably a good idea in general. So while the formatter should of course deal with this correctly, maybe this should also be a linter rule to find potentially error-prone code in the first place ?

cxcxcxcx, 2021-01-29:
I agree this is not good practice and it's a good idea for linter to
complain.

And, I feel a formatter should be robust to such cases. The code itself is
not ambiguous.

If it were not the checker, the problem also sounds very scary.

In fact, I think it works when there are multiple lines of comments. One
line comment is when it fails.

kbrunham-intel, 2026-09-22:
@hzeller I retested the reduced case from this issue with `verible-verilog-format` built from tag `v0.0-4294-gc1d8f5e8`. I could not reproduce the problem (formatter exits 0; output keeps the expected layout / converges). I would suggest closing this issue.

EOL comment between `else if` and the next statement is not glued onto the statement.