chipsalliance/verible #676: Syntax errors due to preprocessor `else macro

msfschaffner, 2021-02-16:
The snipped below fails with the following error due to the `else` preprocessor macro, although the code should be valid. 
```
gen_pok.sv:10:1: syntax error, rejected "`else" (https://github.com/google/verible).
gen_pok.sv:14:1: syntax error, rejected "`endif" (https://github.com/google/verible).
```

Test case:

```systemverilog
`ifndef VERILATOR
module gen_pok #(
// synopsys translate_off
parameter time POK_RDLY = 3us,
parameter time POK_FDLY = 500ns
// synopsys translate_on
) (
  output logic gen_pok_o
);
`else
module gen_pok (
  output logic gen_pok_o
);
`endif

endmodule
```

Verible version:
```
v0.0-808-g1e17daa
Commit	2020-12-14 17:14:57 -0800
Built	2020-12-15T01:35:23Z
```

msfschaffner, 2021-02-16:
b/180420486

Jacob-Levy, 2021-02-21:
% /usr/local/opentitan/tools/verible/0.0-883-g2af116e/bin/verible-verilog-lint gen_pok.sv 
../ip_os/ast/rtl/gen_pok.sv:14:1: syntax error, rejected "`else" (syntax-error).
../ip_os/ast/rtl/gen_pok.sv:25:1: syntax error, rejected "assign" (syntax-error).
../ip_os/ast/rtl/gen_pok.sv:36:1: syntax error, rejected "always" (syntax-error).
../ip_os/ast/rtl/gen_pok.sv:39:3: syntax error, rejected "end" (syntax-error).
../ip_os/ast/rtl/gen_pok.sv:41:3: syntax error, rejected "end" (syntax-error).
../ip_os/ast/rtl/gen_pok.sv:43:3: syntax error, rejected "end" (syntax-error).
../ip_os/ast/rtl/gen_pok.sv:52:1: syntax error, rejected "endmodule" (syntax-error).

Passes: VCS, fusesoc, Verilator, Spyglass, AscentLint

```
`ifndef SYNTHESIS
module gen_pok #(
 parameter time POK_RDLY = 3us,
 parameter time POK_FDLY = 500ns
) (
`else
module gen_pok (
`endif
  output logic gen_pok_o
);
...
endmodule : gen_pok

```

Jacob-Levy, 2021-02-21:
Also fails on the following code...

```
ifndef SYNTHESIS
rng_osc #(
  .RNG_EN_RDLY ( 5us )
) u_rng_osc (
`else
rng_osc u_rng_osc (
`endif
  .vcaon_pok_i ( vcaon_pok_i ),
  .rng_en_i ( rng_en_i ),
  .rng_clk_o ( rng_clk_o )
);  // of u_rng_osc

```
```
rng.sv:38:9: syntax error, rejected "u_rng_osc" (syntax-error).
rng.sv:119:1: syntax error, rejected "endmodule" (syntax-error).
```

fangism, 2021-02-22:
Confirmed and acknowledged. as a known limitation  The current parser implementation lacks a separate preprocessing step, and attempts (best effort) to _parse_ preprocessing directives and construct a syntax tree that includes them, so if a preprocessing conditional does not cleanly surround a "whole" language element grammar construct (e.g. module declaration, module instance), then it will fail like the above.