chipsalliance/verible #690: [kythe] File nodes generated by verible-verilog-kythe-extractor should have empty "language" field.

mglb, 2021-02-26:
## Input test case

`sample_sources/example.sv`:

```systemverilog
module example (input in, output logic out);
  assign in = out;
endmodule
```

`sample_sources.list`:

```
sample_sources/example.sv
```

Shell:

```bash
KYTHE_DIR=/opt/kythe/

OUT_DIR=$PWD/outputs
mkdir -p $OUT_DIR

# Run indexer
verible-verilog-kythe-extractor \
        -print_kythe_facts json \
        -file_list_path 'sample_sources.list' \
        > "$OUT_DIR/entries"

rm -rf $OUT_DIR/graphstore
rm -rf $OUT_DIR/tables

# Convert entries
$KYTHE_DIR/tools/entrystream \
        -read_format=json \
        < "$OUT_DIR/entries" \
    | $KYTHE_DIR/tools/write_entries \
        -graphstore "$OUT_DIR/graphstore"

$KYTHE_DIR/tools/write_tables \
        -graphstore "$OUT_DIR/graphstore" \
        -out "$OUT_DIR/tables"
```

Output:

```
write_entries: 2021/02/26 01:21:03 Wrote 32 entries
2021/02/26 01:21:03 WARNING: proto: file "v1.proto" is already registered
A future release will panic on registration conflicts. See:
https://developers.google.com/protocol-buffers/docs/reference/go/faq#namespace-conflict

2021/02/26 01:21:03 Starting serving pipeline
2021/02/26 01:21:03 Writing partial edges
2021/02/26 01:21:03 Writing complete edges
2021/02/26 01:21:03 Writing EdgeSets
2021/02/26 01:21:03 Writing decoration fragments
2021/02/26 01:21:03 Writing completed FileDecorations
2021/02/26 01:21:03 Warning: no file set for anchor. fileTicket:[kythe:?path=sample_sources/example.sv] curFile:[kythe:?path=sample_sources/example.sv] fragment:[decoration:{anchor:{ticket:"kythe:?lang=verilog?path=sample_sources/example.sv#QDIyOjI0Iw%3D%3D"  start_offset:22  end_offset:24}  kind:"/kythe/edge/defines/binding"  target:"kythe:?lang=verilog?path=sample_sources/example.sv#Li9zYW1wbGVfc291cmNlcy9leGFtcGxlLnN2I2V4YW1wbGUjaW4j"}  target:{ticket:"kythe:?lang=verilog?path=sample_sources/example.sv#Li9zYW1wbGVfc291cmNlcy9leGFtcGxlLnN2I2V4YW1wbGUjaW4j"  fact:{name:"/kythe/complete"  value:"definition"}  fact:{name:"/kythe/node/kind"  value:"variable"}}]
2021/02/26 01:21:03 Warning: no file set for anchor. fileTicket:[kythe:?path=sample_sources/example.sv] curFile:[kythe:?path=sample_sources/example.sv] fragment:[decoration:{anchor:{ticket:"kythe:?lang=verilog?path=sample_sources/example.sv#QDM5OjQyIw%3D%3D"  start_offset:39  end_offset:42}  kind:"/kythe/edge/defines/binding"  target:"kythe:?lang=verilog?path=sample_sources/example.sv#Li9zYW1wbGVfc291cmNlcy9leGFtcGxlLnN2I2V4YW1wbGUjb3V0Iw%3D%3D"}  target:{ticket:"kythe:?lang=verilog?path=sample_sources/example.sv#Li9zYW1wbGVfc291cmNlcy9leGFtcGxlLnN2I2V4YW1wbGUjb3V0Iw%3D%3D"  fact:{name:"/kythe/complete"  value:"definition"}  fact:{name:"/kythe/node/kind"  value:"variable"}}]
2021/02/26 01:21:03 Warning: no file set for anchor. fileTicket:[kythe:?path=sample_sources/example.sv] curFile:[kythe:?path=sample_sources/example.sv] fragment:[decoration:{anchor:{ticket:"kythe:?lang=verilog?path=sample_sources/example.sv#QDU0OjU2Iw%3D%3D"  start_offset:54  end_offset:56}  kind:"/kythe/edge/ref"  target:"kythe:?lang=verilog?path=sample_sources/example.sv#Li9zYW1wbGVfc291cmNlcy9leGFtcGxlLnN2I2V4YW1wbGUjaW4j"}  target:{ticket:"kythe:?lang=verilog?path=sample_sources/example.sv#Li9zYW1wbGVfc291cmNlcy9leGFtcGxlLnN2I2V4YW1wbGUjaW4j"  fact:{name:"/kythe/complete"  value:"definition"}  fact:{name:"/kythe/node/kind"  value:"variable"}}]
2021/02/26 01:21:03 Warning: no file set for anchor. fileTicket:[kythe:?path=sample_sources/example.sv] curFile:[kythe:?path=sample_sources/example.sv] fragment:[decoration:{anchor:{ticket:"kythe:?lang=verilog?path=sample_sources/example.sv#QDU5OjYyIw%3D%3D"  start_offset:59  end_offset:62}  kind:"/kythe/edge/ref"  target:"kythe:?lang=verilog?path=sample_sources/example.sv#Li9zYW1wbGVfc291cmNlcy9leGFtcGxlLnN2I2V4YW1wbGUjb3V0Iw%3D%3D"}  target:{ticket:"kythe:?lang=verilog?path=sample_sources/example.sv#Li9zYW1wbGVfc291cmNlcy9leGFtcGxlLnN2I2V4YW1wbGUjb3V0Iw%3D%3D"  fact:{name:"/kythe/complete"  value:"definition"}  fact:{name:"/kythe/node/kind"  value:"variable"}}]
2021/02/26 01:21:03 Warning: no file set for anchor. fileTicket:[kythe:?path=sample_sources/example.sv] curFile:[kythe:?path=sample_sources/example.sv] fragment:[decoration:{anchor:{ticket:"kythe:?lang=verilog?path=sample_sources/example.sv#QDc6MTQj"  start_offset:7  end_offset:14}  kind:"/kythe/edge/defines/binding"  target:"kythe:?lang=verilog?path=sample_sources/example.sv#Li9zYW1wbGVfc291cmNlcy9leGFtcGxlLnN2I2V4YW1wbGUj"}  target:{ticket:"kythe:?lang=verilog?path=sample_sources/example.sv#Li9zYW1wbGVfc291cmNlcy9leGFtcGxlLnN2I2V4YW1wbGUj"  fact:{name:"/kythe/complete"  value:"definition"}  fact:{name:"/kythe/node/kind"  value:"record"}  fact:{name:"/kythe/subkind"  value:"module"}}]
2021/02/26 01:21:03 Writing CrossReferences
```

(full script for reproducing and testing the problem below)

## Describe what is wrong or missing
[... 248 lines cut ...]
    request "http://$SERVER_IP/decorations" \
        '{' \
            '"location": {' \
                '"ticket":"kythe:?lang=verilog?path=sample_sources/example.sv"' \
            '},' \
            '"source_text":true,' \
            '"references":true,' \
            '"target_definitions":true' \
        '}'

    kill $SERVER_PID
}

echo
echo '*** TEST: ORIGINAL ENTRIES ***'
echo
convert_and_test

jq 'if .source.signature == "" then . | .source.language="" else . end' $OUT_DIR/entries | sponge $OUT_DIR/entries

echo
echo '*** TEST: LANGUAGE REMOVED FROM FILE NODES***'
echo
convert_and_test
```

tgorochowik, 2021-02-26:
CC @ivan444

ivan444, 2021-04-01:
Thanks for a detailed report!

https://github.com/google/verible/pull/729 should fix this. I used your script for testing. Super useful!

PR #729 (2021-04-01): Set Kythe's vname.language to an empty string for the file nodes
Closes #690
closed 2021-04-01 by hzeller