chipsalliance/verible #691: Add integration tests for verible-verilog-kythe-indexer

mglb, 2021-02-26:
Since `verible-verilog-kythe-indexer` output is intended to be passed to Kythe tools, it would be nice to test that the tools accept what has been produced. Things that could be tested:

* Generating Kythe tables from a `verible-verilog-kythe-indexer`'s "entries" file and looking for warnings/errors.
* Starting Kythe server, sending a few requests and checking the replies.

Example test can be found in issue https://github.com/google/verible/issues/690 in the "Testing script" section.

ivan444, 2021-02-27:
What's the plan for making the integration test robust? Doing any response expectation limits the variation of the SystemVerilog Kythe facts extraction.

Kythe's Verification Tests do a good job at ensuring the facts we produce are correct (from the perspectice of Kythe schema, not what the tools need). How about checking that the indexer produced at least N facts, compared to expecting the exact replies?

mglb, 2021-03-08:
> How about checking that the indexer produced at least N facts, compared to expecting the exact replies?

That's something I had in mind. Also: checking that request for known indexed file doesn't return error, that it returns non-empty string in `source_text` field, etc. This should be reasonably robust.