chipsalliance/verible #693: Syntax errors due to streaming `with` operator

udinator, 2021-03-03:
The  SV linter is throwing unexpected errors on the `with` construct used in streaming operations, even though this construct is part of the LRM.
Example error message:
`../src/lowrisc_dv_mem_bkdr_if_0/sram_scrambler_pkg.sv:58:28: syntax error, rejected "with" (https://github.com/google/verible).`

Example test case:
```systemverilog
module test;
  initial begin
    logic unpacked_arr [16] = '{0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1};
    logic [3:0] nibble_arr[4];
    for (int i = 0; i < 4; i++) begin
      nibble_arr[i] = {<< {unpacked_arr with [i*4 +: 4]}};
    end
  end
endmodule
```

udinator, 2021-03-03:
b/181792412

hzeller, 2021-03-03:
Context: this is LRM 14.4.14
And yes, confirmed, it is indeed missing.

hzeller, 2021-03-03:
Also caught by [sv-tests](https://symbiflow.github.io/sv-tests-results/#Verible|11.4.14.4|dynamic_array_unpack_stream_with)

hzeller, 2021-03-04:
More context: this was [indeed thought of](https://github.com/google/verible/blob/35680ebb3eed3ebc0200610252f857fa8e381510/verilog/parser/verilog.y#L3223-L3225) but not added initially due to parser generation conflicts.
(just collecting this here as context for whoever will be working on this)