chipsalliance/verible #716: How to use Verible in a new language?

WenhMa, 2021-03-26:
If I want to develop a SystemVerilog liked language that only offers some simple new features, how should I start with Verible?

hzeller, 2021-03-26:
For supporting new languages, there is already _some_ infrastructure in place to provide a framework to do it. Right now, it is used with exactly one supported language (SystemVerilog), so we might see what else needs to go into that.

There is the [`common/`](https://github.com/google/verible/tree/master/common) directory that contains language agnostic parts (the [README](https://github.com/google/verible/blob/master/common/README.md) there give an high-level overview). It introduces concepts for parsing, tokenization etc.

Then there are subdirectories from the toplevel for the different languages; currently the only supported language is [`verilog/`](https://github.com/google/verible/tree/master/verilog), which then has its own [parser/](https://github.com/google/verible/tree/master/verilog/parser) subdirectory. The README there is helpful to get started on the concepts used there.

So with this in mind, I'd suggest going through the READMEs (that are in most sub-directories) that can help you finding all the components and ideas needed to support your new language. Then come back here for specific questions that we can help answer here.
 
If @fangism is around, he can probably also comment on this as he has designed the language-independent infrastructure.

fangism, 2021-03-26:
Can you elaborate on what features you want to develop for your language?

WenhMa, 2021-04-02:
I'm trying to develop a new SystemVerilog with some simple new features. Perhaps we can call it 'Improved SystemVerilog'(isv). It is only a code generator. The main focus of 'isv' is providing some user-friendly features to simplify the SystemVerilog code. The most important thing is that these features will not introduce too many intermediate variables which makes it difficult to verify and implement the design. Some possible features are demonstrated below as demos.

1. Define signals like c-style.
    `logic [7:0] a;    ->      logic [8] a;`
2. Support namespace like python.
    If there are two modules called 'moduleAa' and 'moduleBb' in file 'ModuleLibrary.isv'. Use them like:
```
    import ModuleLibrary as mlib;
    mlib.moduleAa#(...)(...);
    mlib.moduleBb#(...)(...);
```
3. ' if<> ' for parameter based conditional statement. 
    If a conditional statement is only used for circuit generating that only parameters are included in the statement, use a new operator.
```
   parameter AA = 1;
   if<AA>0> ......
   else           ......
   always_comb begin
     if<AA>0>.....
     else  .....
   end
```
These three features are only part of the 'Improved SystemVerilog'. I think it's better to provide an infrastructure that we can discuss and try new features for future versions of SystemVerilog. Do you have any suggestions?

hzeller, 2021-04-02:
it sounds like you'd need to extend the lexer and parser of the current SystemVerilog implementation, but in a not backward compatible way, but sufficiently overlapping that it is worthwhile starting with the current verilog.

I suggest to start out in your local experiments with a branch in which you modify the lexer and parser according to accept your new language with new tests that test for every single new language feature, while also making sure not to break old features if possible.

Once that is worked out we can see if it is possible to consolidate that. We do have dialect support on our TODO list (#141), but details have not been hashed out yet. But maybe this could be made a dialect iff it is not too different from SystemVerilog.