chipsalliance/verible #717: Explicitly typed structure literals are marked as syntax error

zegervdv, 2021-03-26:
**Describe the bug**

Verible marks explicit types on structure literals as a syntax error.

**To Reproduce**

```systemverilog
module foobar;
  always_comb begin
    foo = t_foo'{
      bar: '1,
      boo: '0
    };
  end
endmodule
```

Include any options used.

**Actual behavior:**

```
test.sv:3:16: syntax error, rejected "'{" (syntax-error).
```
It works fine without the explicit `t_foo` type.

**Expected behavior**

Section 5.10 of the LRM specifies that structure literals may have their data type defined, sometimes it is needed to remove ambiguity.

snsokolov, 2021-03-29:
Confirming that the syntax is valid. The following code (more complete example) compiles in other simulators.

```
module foobar;
  typedef struct packed {
    logic bar;
    logic boo;
  } t_foo;
  t_foo foo;
  always_comb begin
    foo = t_foo'{
      bar: '1,
      boo: '0
    };
  end
endmodule
```

j-marjanovic, 2021-04-22:
Basically +1, but I would just like to add that this also happens when the fields of the structure literal are not named, e.g.:

```SystemVerilog
module struct_test;

  typedef struct {bit [31:0] addr;} t_read_cmd;

  t_read_cmd read_cmd_queue[$];

  task start_read(bit [31:0] addr);
    read_cmd_queue.push_back(t_read_cmd'{addr});
  endtask
endmodule
```

* version: v0.0-1144-gbdaa5bd
* result: `tmp1.sv: tmp1.sv:8:40: syntax error, rejected "'{" (syntax-error).`

PR #1127 (2022-01-05): Add explicit typed structure literals
A naive attempt to fix  #717.

I have introduced a new node `kAssignmentPatternExpression` to handle the struct literals (inspired by the parser from [tree-sitter-verilog](https://github.com/tree-sitter/tree-sitter-verilog)).

I have zero experience with yacc so I've probably overlooked things. 
Any comments or feedback would be welcome.
commit eef8c222 (2022-01-05): Add explicit typed structure literals

fixes #717
closed 2022-01-05 by hzeller