chipsalliance/verible #725: [kythe] Modules instances should link to the module definition

suzizecat, 2021-03-31:
## Input test case

Try to reduce the test down to a single demonstrative file.

```systemverilog
module child (
	input  i,
	output o
);
assign o = i;
endmodule

module parent ();

logic i_signal;
logic o_signal;

child inst(.i(i_signal),.o(o_signal));
endmodule
```

## Describe what is wrong or missing

In the Kythe tree, there is no direct link between `inst` and `child`.
As far as I understood, this could be done through an _instanciates_ relationship

hzeller, 2021-03-31:
Adding @ivan444 who knows a lot more detail about the Kythe implementation, and can probably comment.

We are in the process of internally revamping an implementation of the symbol table; so far we only use it for the [project tool]( https://github.com/google/verible/tree/master/verilog/tools/project), while the kythe-indexer is still using an earlier form of that effort, so there is a lot more to be expected; currently we are focusing our efforts on other parts, so if you'd like to get involved, having a look at the Kythe implementation and the symbol table will certainly be some worthwhile places to start.

ivan444, 2021-04-01:
Hi!

`instantiates`[1] is used for specializations, like C++ templates. We didn't design SystemVerilog modules to behave like templated C++ classes, but more like a regular C++ classes (we produce an instance[2], not an instantiation[1]).

Kythe verification tests which shows which Kythe facts we output for module instances: https://github.com/google/verible/blob/master/verilog/tools/kythe/testdata/module_extra_test.sv

What use-case our design breaks? (we can revisit if needed)

[1] https://kythe.io/docs/schema/#instantiates
[2] https://kythe.io/docs/schema/#definesbinding

suzizecat, 2021-04-01:
Hi !

On a theoretical level, I would argue that a module instantiation (as in the action to create an instance) is actually a specification as long as you have parameters. However, that's not my issue here 😉 

My issue is that an instance (here `inst`) have no link to its type (`child`) in the Kythe output. Hence it is tedious to reliably get from `#parent#inst` to `#child` using only the information provided by Kythe. 

I suggested `instanciates`  as it was the one most fitting to my unknowing eyes, I actually don't really care what exactly is the fact type.
It might also be worth considering linking `#parent#inst` to `#parent#child` which is already properly referring to `#child` even though I don't have a proper use case in mind for this.

ivan444, 2021-04-01:
That's a good point. I need to see how that fits the Kythe schema. At the moment I haven't seen good examples.