chipsalliance/verible #726: Consider making a pure flex parser for potential distribution of the pre-generated file

hzeller, 2021-04-01:
Currently, we're using flex to generate the lexer. This has all kinds of portability issues on platforms not Unix (#307 #260).

To minimize dependencies on the generated code, one option would be to check it into the repository and potentially update only on change.

The current c++ generated flex code however also requires another header to interface, which comes with the particular flex installation. It might be worthwhile investigating to just generate a C lexer, since we have our interfacing c++ wrapper around it anyway. Once that is done, the 'distribute the generated code' might also be easier.

mkschreder, 2023-11-11:
The fact that flex and bison by default do not generate pure reentrant C code is sickening 🤮

hzeller, 2023-11-13:
We're using the C++ wrapper, so this is not one of the problems; they are reentrant.
(Only if you use the C version without requesting reentrance this will be a problem; but not used in this project).