chipsalliance/verible #743: can Token Text (Signal name, port name, parameter name definitions) be available in JSON output?

SingaDK, 2021-04-05:
Hi,
I want to work on signal dependencies based on the CST JSON output of SystemVerilog modules.
It would be really helpful if I can get the defined port names, signal, wire, reg names in JSON output itself.
It would be hard to get the token text by using the byte converted source file with byte offset for getting information about one particular signal or port.
Unfortunately current --export_json flag doesn't have a separate field for the original defined.
Is there a way to have the JSON which also includes the port, module names in it?
Thanks in advance

hzeller, 2021-04-05:
Beyond looking at the lower-level syntax tree, these requirements might be better served with some additional outputs from the higher-level symbol-table frontend ( https://github.com/google/verible/tree/master/verilog/tools/project - currently just use to provide some dependency reports but the internal representation contains a lot more to tap into) or the Kythe indexer ( https://github.com/google/verible/tree/master/verilog/tools/kythe ) which currently extracts dependencies with its own implementation of a symbol table, though these tools eventually will both use the same.

Did you consider these as well ?

SingaDK, 2021-04-05:
Hi,
Many thanks for your quick prompt.
I really wanted to try tools/project or the tools/kythe. But I am getting "terminate called after throwing an instance of 'std::__ios_failure'
  what():  basic_filebuf::underflow error reading the file: iostream error
Aborted (core dumped)" error when using both tools. 
Since I am a beginner in this field, I wasn't sure what causes this error. Any pointer towards it will be highly appreciated.
As you correctly mentioned, a symbol table will be of high advantage for my requirement but I am unable to get it out due to ios_failure error.
Thanks in advance.

hzeller, 2021-04-05:
Can you post the command line you were using to invoke the tools ?

SingaDK, 2021-04-05:
verible-verilog-kythe-extractor --printextraction --file_list_path "/home/singa/Desktop/ibex/ibex/rtl/"
verible-verilog-project file-deps --file_list_path "/home/singa/Desktop/ibex/ibex/rtl/"

and gcc version is gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0

hzeller, 2021-04-06:
Ah. the `--file_list_path` takes a file that lists all the file-names to be looked at. You are giving it a directory name, which it then is confused (the error message should be better in that case...)

hzeller, 2021-04-06:
Ok, improved error message in this case.

So what you need is a file that has a list of files you want to analyze, and then pass that to the tool in question
```
ls rtl/*.sv > /tmp/filelist
verible-verilog-kythe-extractor --printextraction --file_list_path=/tmp/filelist
```

If you are using bash, you can do that also on-the-fly
```
verible-verilog-kythe-extractor --printextraction --file_list_path=<(ls rtl/*.sv)
```

SingaDK, 2021-04-07:
It works like a charm now, really thanks a lot for the prompt replies. Really loving this project!!

As for the initial question,
Is it really possible to include the signal name which is printed in --printtree CST and also in Pretty print_tree.py? but why it's not printed with the export_json flag in the JSON file?
My aim is to somehow get, what are the signals driven by a particular signal or a signal fan-out. I want to automate this by getting the parse tree and more easily if it's CST's JSON for a source file. 
Is it possible to tweak the JSON export to include the defined port or signal name too?

hzeller, 2021-04-12:
Passing on to Mariusz who implemented the Python interface.

mglb, 2021-04-14:
Hi. Exporting signal names in a JSON is not possible in current version of Verible, but it can be implemented.
The intended way to get this info currently is (as you already wrote) to read a source file into an array and extract a string from it using offsets found in a JSON. That's what `print_tree.py` does.

I'll look into the code later, it seems to be quick and easy change.

PR #816 (2021-05-25): Include tokens text in JSON output
Adds `text` field containing token text to all token nodes except those representing operators and keywords.

Closes #743
closed 2021-05-25 by tgorochowik