chipsalliance/verible #748: Format OpenTitan codebase with Verible SystemVerilog code formatter

ldalek-antmicro, 2021-04-05:
**Use Verible formatter to format OpenTitan (https://github.com/lowRISC/opentitan) codebase.**

Modify formatter so it complies with [lowRISC Verilog Coding Style](https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md) and use it with OpenTitan codebase.

**Current formatting results**
https://github.com/ldalek-antmicro/opentitan/commit/034e844c171ae86c475311f3e561fb6eb7efdc6e

**Missing features/bugs**

1. Failing files (syntax error):
```
./hw/dv/sv/cip_lib/seq_lib/cip_base_vseq.sv
./hw/dv/sv/cip_lib/seq_lib/cip_base_vseq__tl_errors.svh
./hw/dv/sv/tl_agent/tl_seq_item.sv
./hw/ip/edn/dv/env/edn_env_cfg.sv
./hw/ip/flash_ctrl/dv/env/seq_lib/flash_ctrl_base_vseq.sv
./hw/ip/gpio/dv/env/gpio_scoreboard.sv
```

**Allow list**
Start with an allow list with fileset that complies with formatting style guide.
1. [hw/dv/sv/alert_esc_agent/alert_esc_base_monitor.sv](https://github.com/ldalek-antmicro/opentitan/commit/034e844c171ae86c475311f3e561fb6eb7efdc6e#diff-e5dda8c8fc5a71bc4977db061f4b774fc1ef3d147152810bcaf66a156a26821a)
2. [hw/dv/sv/alert_esc_agent/alert_esc_probe_if.sv](https://github.com/ldalek-antmicro/opentitan/commit/034e844c171ae86c475311f3e561fb6eb7efdc6e#diff-e6f7b58b2db1180e7fc86f5700246496017c5a22cb51b47e32483d9394e08fe2)

hzeller, 2021-04-05:
(If there are parse errors, there is one open pull request #697 that works on one parsing issue, not sure if this is one that is problematic here)

hzeller, 2021-04-06:
Also as we had looked at before: relevant are these pull request, in which OpenTitan folks commented on the format output, which helps give us insight which is most important.

https://github.com/lowRISC/ibex/pull/1215
https://github.com/msfschaffner/opentitan/pull/7 (older)

Might be good if we can pair these comments with existing issues mentioned in https://github.com/google/verible/labels/formatter and increase priority as they are clearly needed.

hzeller, 2021-04-06:
In this run I did a while back https://github.com/hzeller/opentitan/commit/e28c2f62f61638c867c4feb0cd5fef1041f30dc7
David also added some links to existing bugs.