chipsalliance/verible #753: "slew" detected as reserved word (?

gasrodriguez, 2021-04-14:
**Describe the bug**

Cannot use the word "slew" as module name nor port or signal name. 

**To Reproduce**

Run verible-verilog-syntax for the following snippet, using slew in each position one at a time:

```systemverilog
module slew (
    input logic slew
);
logic slew;
endmodule;
```

Include any options used.

**Actual behavior:**

Rejects valid code.

**Expected behavior**

It looks as if the parsers thinks is a reserved word, but I [checked ](https://www.intel.com/content/www/us/en/programmable/quartushelp/17.0/hdl/vlog/vlog_file_reserved_words.htm) and it isn't.

hzeller, 2021-04-14:
Is this one of the AMS keywords @fangism ? Looks like they use it for Analog  filtering (VAMS-LRM Section 4.5).

@gasrodriguez Currently, we're accepting a superset of keywords going beyond SystemVerilog but also covering [mixed-signal descriptions](https://en.wikipedia.org/wiki/Verilog-AMS) - this is one of these.

It is on our TODO list to have dialect support ( #141 ), but didn't get around to it. So in the current stage, I am afraid, you might need to use a different word.

fangism, 2021-04-14:
Confirmed as keyword.

PR #1229 (2022-03-01): Make "slew" a simple SymbolIdentifier.
Right now, it is not used as a keyword anywhere, so not even
a need to make it a KeywordIdentifier.

Fixes #753
commit 9a5c6db6 (2022-03-01): Make "slew" a simple SymbolIdentifier.

Right now, it is not used as a keyword anywhere, so not even
a need to make it a KeywordIdentifier.

Fixes #753
closed 2022-03-01 by hzeller