chipsalliance/verible #758: [Mac-os]: create a brew installable binary

hzeller, 2021-04-14:
We could probably use github packages to provide a brew bottle/tap of sorts ? Details need to be figured out.

hzeller, 2021-04-16:
Paging @speedy-beaver who might be able to give an overview of required steps to get there.

We do releases essentially on every commit, so to not annoy upstream brew with multiple pull requests a day, we should probably have our own brew repository that people can point their brew install to.

hzeller, 2021-04-16:
So looks like this might be possible just with github means
https://brew.sh/2020/11/18/homebrew-tap-with-bottles-uploaded-to-github-releases/

johanvdhaegen, 2021-04-20:
I assume you would like the steps for the end user to be as follows:

```
brew tap google/verible
brew install verible
```

In that case, the homebrew formula should be a in github repo called `homebrew-verible` under the `google` org on github.

The idea would be that commits to the verible repo, would trigger version bump PR in the homebrew-verible repo, which in turn would trigger a bottle rebuild.

mithro, 2021-04-20:
@hzeller - If this is the direction you want to go poke me and I'll get the `google/homebrew-verible` github repository created for you.

hzeller, 2021-04-20:
Yes, Let's make a `google/homebrew-verible` repo!

mithro, 2021-04-20:
I started the process to get that created. Will ping you when it exists.

mithro, 2021-04-22:
https://github.com/google/homebrew-verible now exists and @hzeller has admin control over it.

hzeller, 2021-04-22:
Yes, ready to accept pull requests there.

johanvdhaegen, 2021-04-22:
First PR with homebrew github workflows mentioned in doc referenced [above](https://github.com/google/verible/issues/758#issuecomment-821525068):

https://github.com/google/homebrew-verible/pull/1

hzeller, 2023-05-02:
Homebrew package is done here. 
To automatically update, work continues in #1904