chipsalliance/verible #760: Testing later version of Kythe: new build problems

hzeller, 2021-04-14:
Resolution in #625 was to clamp 0.0.49 to some older version of Kythe. Did a quick test today to see if the latest version `0.0.52` would work, but different issues emerge:

Using this in WORKSPACE
```
http_archive(
    name = "io_kythe",
    sha256 = "dde86b6fa277486665dc83c66aaf9f5f961427e3e417538a09498494c0be0390",
    strip_prefix = "kythe-0.0.52",
    urls = [
        "https://github.com/kythe/kythe/archive/refs/tags/v0.0.52.tar.gz"
    ],
)
```
Stumbles upon some ruby dependency.
<details><summary>bazel build log output</summary>

```
$ bazel build ...
INFO: Repository bazelruby_rules_ruby instantiated at:
  /home/hzeller/src/my/verible/WORKSPACE:152:24: in <toplevel>
  /home/hzeller/.cache/bazel/_bazel_hzeller/fde5431bd58d2dc68f037724f34a8e77/external/io_kythe/setup.bzl:120:10: in kythe_rule_repositories
  /home/hzeller/.cache/bazel/_bazel_hzeller/fde5431bd58d2dc68f037724f34a8e77/external/io_kythe/setup.bzl:7:18: in maybe
Repository rule http_archive defined at:
  /home/hzeller/.cache/bazel/_bazel_hzeller/fde5431bd58d2dc68f037724f34a8e77/external/bazel_tools/tools/build_defs/repo/http.bzl:336:31: in <toplevel>
INFO: Repository 'bazelruby_rules_ruby' used the following cache hits instead of downloading the corresponding file.
 * Hash 'abfc2758cc379e0ff0eb9824e3b507c1633d4c8f99f24735aef63c7180be50f0' for https://github.com/bazelruby/rules_ruby/archive/v0.4.1.zip
If the definition of 'bazelruby_rules_ruby' was updated, verify that the hashes were also updated.
ERROR: An error occurred during the fetch of repository 'bazelruby_rules_ruby':
   Traceback (most recent call last):
        File "/home/hzeller/.cache/bazel/_bazel_hzeller/fde5431bd58d2dc68f037724f34a8e77/external/bazel_tools/tools/build_defs/repo/http.bzl", line 121, column 10, in _http_archive_impl
                patch(ctx)
        File "/home/hzeller/.cache/bazel/_bazel_hzeller/fde5431bd58d2dc68f037724f34a8e77/external/bazel_tools/tools/build_defs/repo/utils.bzl", line 135, column 22, in patch
                ctx.patch(patchfile, strip)
Error in patch: Unable to load package for //third_party:rules_ruby_allow_empty.patch: BUILD file not found in any of the following directories. Add a BUILD file to a directory to mark it as a package.
 - /home/hzeller/src/my/verible/third_party
ERROR: Error fetching repository: Traceback (most recent call last):
        File "/home/hzeller/.cache/bazel/_bazel_hzeller/fde5431bd58d2dc68f037724f34a8e77/external/bazel_tools/tools/build_defs/repo/http.bzl", line 121, column 10, in _http_archive_impl
                patch(ctx)
        File "/home/hzeller/.cache/bazel/_bazel_hzeller/fde5431bd58d2dc68f037724f34a8e77/external/bazel_tools/tools/build_defs/repo/utils.bzl", line 135, column 22, in patch
                ctx.patch(patchfile, strip)
Error in patch: Unable to load package for //third_party:rules_ruby_allow_empty.patch: BUILD file not found in any of the following directories. Add a BUILD file to a directory to mark it as a package.
 - /home/hzeller/src/my/verible/third_party
INFO: Repository build_bazel_rules_nodejs instantiated at:
  /home/hzeller/src/my/verible/WORKSPACE:152:24: in <toplevel>
  /home/hzeller/.cache/bazel/_bazel_hzeller/fde5431bd58d2dc68f037724f34a8e77/external/io_kythe/setup.bzl:84:10: in kythe_rule_repositories
  /home/hzeller/.cache/bazel/_bazel_hzeller/fde5431bd58d2dc68f037724f34a8e77/external/io_kythe/setup.bzl:7:18: in maybe
Repository rule http_archive defined at:
  /home/hzeller/.cache/bazel/_bazel_hzeller/fde5431bd58d2dc68f037724f34a8e77/external/bazel_tools/tools/build_defs/repo/http.bzl:336:31: in <toplevel>
INFO: Repository io_bazel_rules_rust instantiated at:
  /home/hzeller/src/my/verible/WORKSPACE:152:24: in <toplevel>
  /home/hzeller/.cache/bazel/_bazel_hzeller/fde5431bd58d2dc68f037724f34a8e77/external/io_kythe/setup.bzl:110:10: in kythe_rule_repositories
  /home/hzeller/.cache/bazel/_bazel_hzeller/fde5431bd58d2dc68f037724f34a8e77/external/io_kythe/setup.bzl:7:18: in maybe
Repository rule http_archive defined at:
  /home/hzeller/.cache/bazel/_bazel_hzeller/fde5431bd58d2dc68f037724f34a8e77/external/bazel_tools/tools/build_defs/repo/http.bzl:336:31: in <toplevel>
ERROR: no such package '@bazelruby_rules_ruby//ruby': Unable to load package for //third_party:rules_ruby_allow_empty.patch: BUILD file not found in any of the following directories. Add a BUILD file to a directory to mark it as a package.
```

</details>

This probably all happens while processing the [setup.bzl](https://github.com/kythe/kythe/blob/master/setup.bzl) in Kythe.

Passing on to @ivan444

hzeller, 2021-04-14:
This prevents us from moving to Bazel 4.0.0 ( #759 attempt ), because the old `vname.bzl` uses a syntax that is more thoroughly checked with the newer version.

ivan444, 2021-04-15:
We can get rid of Kythe Bazel dependency if we use precompiled version of the Kythe verification binary. What do you think? (please take a look at b/171679296)

PR #772 (2021-04-20): [kythe] Stop depending on Kythe at the build time
The only remaining dependency is proto schema which was directly imported to the project's third_party directory.

Closes #760
closed 2021-04-20 by hzeller