chipsalliance/verible #765: Move to use C++17 and with it, embrace std::filesystem

hzeller, 2021-04-19:
- [ ] Evaluate if we can move to c++17, by checking all the (sometimes pretty old) distributions we support when building.
- [ ] Are there inconveniences that we need to factor in ? For instance do we need to tell users to add `--cxxopts=-std=c++17`, or is that the default already with using a current bazel (4.0.0). If it is the default with 4.0.0 (current version bazel) but not 3.7 which we use for CI currently, then we might first need to await the Kythe dependency reduction (which now pins us on 3.7)
  - [ ] Adapt CI to use C++17 enabled compiler if not already
  - [ ] Add possible documentation updates if special `cxxopts` are needed when building (hopefully not).
- [ ] Replace all unix-specific code with std::filesystem
   - [ ] There is at least one function [`UpwardFileSearch()`](https://github.com/google/verible/blob/6ef4938d0a800b93ebf9fa7fdf692c4ed06c0471/common/util/file_util.h#L56-L60) that needs a unit test first. Possibly more.
   - [ ] To use `std::filesystem` in the past, there was the inconvenience to have to add `-lstdc++fs`. This might probably be possible to add to the `copts` in the BUILD file, but we need to make sure this works on all `gcc` and `clang` a well as on all used compiler versions of these (because it seems to be impossible in bazel to change `copts` in BUILD files depending on toolchain used).

This issue will probably the aggregation point for multiple submits until we have replaced

hzeller, 2021-04-23:
In #789 an implementation to used `std::filesystem` has been added, but _only_ enabled for Windows right now as the release builds (in [`releasing/`](https://github.com/google/verible/tree/master/releasing)) sometimes run on such old platforms that they don't have new enough compilers yet.

TODO: install the necessary compilers there (backports?) and always [enable std::filesystem]( https://github.com/google/verible/blob/495dda72923d1a13422a6f3a3fd131ed3b1fff37/common/util/file_util.cc#L44) for all platforms.

Once that is done, the legacy implementation can be removed.

hzeller, 2021-04-23:
Assigning to Tim to figure out the release compiler situation.

mithro, 2021-04-23:
I'm unclear what you want me to do? Figure out a way to get newer GCC installed on the older CentOS?

hzeller, 2021-04-23:
Correct. I suspect there are backport channels of sorts that have newer compilers available. Since you wrangled these issues before, you might see quickly what needs to be done.

mithro, 2021-04-24:
If the only problem is CentOS, then it is pretty easy. CentOS has these things called `devtoolset` -- see https://www.softwarecollections.org/en/scls/rhscl/devtoolset-8/ and the lines @ https://github.com/google/verible/blob/master/releasing/centos/6/bazel.dockerstage#L11 and https://github.com/google/verible/blob/master/releasing/centos/6/bazel.dockerstage#L25

hzeller, 2021-04-24:
Also the Ubuntu Xenial and Bionic were affected https://github.com/google/verible/runs/2415978347

mithro, 2021-04-24:
You need to update https://github.com/google/verible/blob/master/releasing/ubuntu/xenial/compiler.dockerstage and add the equivalent for Bionic.

PR #805 (2021-05-03): Use std::filesystem & GCC 9
Closes #765. Use `std::filesystem` on all platforms (and with it C++17).

To achieve this, I updated the compiler to GCC 9 on all platforms, which does not need an extra link flag to bring `std::filesystem`. It could be done with GCC 8 by adding the link flag `-lstdc++fs`. However, of all the platforms that need a "non default compiler", only CentOS 8 has gcc 8, so the cost to bring GCC 9 is small. Since GCC 9 is 2 years old, I opted for the newer compiler and leaving the compile flag as is.
commit a152f6f5 (2021-05-03): Merge pull request #805 from corco/std-filesystem-gcc-9

Use std::filesystem & GCC 9
Fixes #765
closed 2021-05-03 by hzeller