chipsalliance/verible #774: Add ability to output a problem_matcher.json file

mithro, 2021-04-20:
**Summary**

`problem_matcher.json` is a file format used by both Visual Studio Code and GitHub actions to enable the showing of annotations in the IDE / GitHub Pull Request (annotations).

More information on the `problem_matcher.json` format can be found at;
 - https://vscode-docs.readthedocs.io/en/stable/editor/tasks/#defining-a-problem-matcher
 - https://github.com/actions/toolkit/blob/master/docs/problem-matchers.md
 - https://github.com/actions/toolkit/blob/master/docs/commands.md#problem-matchers

First, the verible tools will need to write the matchers to a `.json` file. Then when running on GitHub Actions, they will need to output;
```
::add-matcher::<path to .json>
```

FYI -- There is one slight complication that if Verible is running in a docker container on GitHub actions the path needs to be accessible to the runner outside the container. The [`SymbiFlow/actions/checks@main`](https://github.com/SymbiFlow/actions/tree/main/checks) action uses [the following Python code for that](https://github.com/SymbiFlow/actions/blob/main/checks/checks.py#L468-L505);
```python
def matcher_add():
    """
    See https://github.com/actions/toolkit/blob/main/docs/commands.md#problem-matchers
    The matcher json must be available in the **current** workspace.
    See https://github.community/t/problem-matcher-not-found-in-docker-action/16814/2
    Hence we read the matcher data from the docker container and then write it
    out into the workspace before outputting the `::add-matcher` command.
    """
    if not OUTPUT_ANNOTATIONS:
        return

    # Read in the problem_matcher.json data
    infile = __path__ / 'problem_matcher.json'
    with open(infile) as f:
        data = f.read()

    assert MATCHER_OWNER in data, (MATCHER_OWNER, data)

    # Write out the problem_matcher.json data to a file in the local directory.
    fname = 'symbiflow_checks.json'
    inside_docker = pathlib.Path('/github/workflow')
    if inside_docker.exists():
        logging.debug('Running inside GitHub Action docker container!')
        outside_docker = pathlib.Path(os.environ['RUNNER_TEMP']) / '_github_workflow'
    else:
        logging.debug('Running outside docker container!')
        inside_docker = pathlib.Path(tempfile.gettempdir())
        outside_docker = inside_docker

    logging.debug('Matcher will be written to: %s', inside_docker)
    logging.debug('Matcher will be found at: %s', outside_docker)

    with open(inside_docker / fname, 'w') as f:
        f.write(data)

    print('::add-matcher::{}'.format(outside_docker / fname))
```

**Test cases and examples**

The more examples you provide, the easier it will be for someone to come along
and implement it.

```systemverilog
// This example should be diagnosed.
```

```systemverilog
// This variant should NOT be diagnosed (negative test case)
```

**Proposal**

[optional] Suggest approach or include pseudo code for how one might implement
the rule.

**Additional context**

List citations to published style guides, or include relevant excerpts.

hzeller, 2021-04-20:
Tom, assigning to you for now as you were exploring that. Thanks to Tim's work and documentation above, we now have a clear picture what to work towards (also see #773 as an example).

mithro, 2021-04-20:
BTW - If we can do without the Python wrapper script (and potentially even the docker container if we have Mac / Win / Linux support), that is probably a better idea?