chipsalliance/verible #775: Formatter can't handle this (weird) instanciation

gasrodriguez, 2021-04-20:
**Test case**

```systemverilog
module test ();

  submodule dut (
      .clk()
      // comment
      ,
      .rst()
  );

endmodule
```

There is a comment line between the clock and the comma, and that breaks something in the formmater. AFAIK that is valid SV code. We re-arranged the code so is not an issue, but wanted to submit it just in case there is something that can be fixed there :)

**Actual output**

Formatter fails without any output.

hzeller, 2021-04-22:
Thanks for the report!

As first step for documenting what is happening, ran debug binary with gdb:

```
$ bazel build -c dbg verilog/tools/formatter:verible-verilog-format
$ gdb --args bazel-bin/verilog/tools/formatter/verible-verilog-format /tmp/test775.sv
run
```

This is the stacktrace when invoking it with the file
<details><summary>Crash stacktrace</summary>

```
Program received signal SIGSEGV, Segmentation fault.
0x00005555556ad238 in verible::SymbolCastToNode (symbol=...) at common/text/tree_utils.cc:111
111       CHECK_EQ(symbol.Kind(), SymbolKind::kNode)
(gdb) bt
#0  0x00005555556ad238 in verible::SymbolCastToNode (symbol=...) at common/text/tree_utils.cc:111
#1  0x000055555559c1db in verilog::formatter::IgnoreWithinActualNamedPortPartitionGroup (partition=...) at verilog/formatting/align.cc:148
#2  0x00005555555aeb09 in std::__invoke_impl<bool, bool (*&)(verible::VectorTree<verible::UnwrappedLine> const&), verible::VectorTree<verible::UnwrappedLine> const&> (
    __f=@0x7fffffffc720: 0x55555559c153 <verilog::formatter::IgnoreWithinActualNamedPortPartitionGroup(verible::TokenPartitionTree const&)>)
    at /usr/include/c++/10/bits/invoke.h:60
#3  0x00005555555ac043 in std::__invoke_r<bool, bool (*&)(verible::VectorTree<verible::UnwrappedLine> const&), verible::VectorTree<verible::UnwrappedLine> const&> (
    __fn=@0x7fffffffc720: 0x55555559c153 <verilog::formatter::IgnoreWithinActualNamedPortPartitionGroup(verible::TokenPartitionTree const&)>)
    at /usr/include/c++/10/bits/invoke.h:113
#4  0x00005555555a970f in std::_Function_handler<bool (verible::VectorTree<verible::UnwrappedLine> const&), bool (*)(verible::VectorTree<verible::UnwrappedLine> const&)>::_M_invoke(std::_Any_data const&, verible::VectorTree<verible::UnwrappedLine> const&) (__functor=..., __args#0=...) at /usr/include/c++/10/bits/std_function.h:291
#5  0x00005555555d9945 in std::function<bool (verible::VectorTree<verible::UnwrappedLine> const&)>::operator()(verible::VectorTree<verible::UnwrappedLine> const&) const (
    this=0x7fffffffc720, __args#0=...) at /usr/include/c++/10/bits/std_function.h:622
#6  0x00005555555d3a58 in verible::FilterAlignablePartitions(verible::container_iterator_range<__gnu_cxx::__normal_iterator<verible::VectorTree<verible::UnwrappedLine>*, std::vector<verible::VectorTree<verible::UnwrappedLine>, std::allocator<verible::VectorTree<verible::UnwrappedLine> > > > > const&, std::function<bool (verible::VectorTree<verible::UnwrappedLine> const&)> const&) (range=..., ignore_partition_predicate=...) at common/formatting/align.cc:667
#7  0x000055555559d7bf in verilog::formatter::ExtractAlignablePartitionGroups(const std::function<std::vector<verible::TaggedTokenPartitionRange, std::allocator<verible::TaggedTokenPartitionRange> >(const verible::container_iterator_range<__gnu_cxx::__normal_iterator<verible::VectorTree<verible::UnwrappedLine>*, std::vector<verible::VectorTree<verible::UnwrappedLine>, std::allocator<verible::VectorTree<verible::UnwrappedLine> > > > >&)> &, const verible::IgnoreAlignmentRowPredicate &, const verible::TokenPartitionRange &, const verilog::formatter::FormatStyle &) (group_extractor=..., ignore_group_predicate=..., full_range=..., vstyle=...) at verilog/formatting/align.cc:1135
#8  0x000055555559db19 in verilog::formatter::AlignActualNamedPorts (full_range=..., vstyle=...) at verilog/formatting/align.cc:1166
#9  0x00005555555aecd9 in std::__invoke_impl<std::vector<verible::AlignablePartitionGroup, std::allocator<verible::AlignablePartitionGroup> >, std::vector<verible::AlignablePartitionGroup, std::allocator<verible::AlignablePartitionGroup> > (*&)(verible::container_iterator_range<__gnu_cxx::__normal_iterator<verible::VectorTree<verible::UnwrappedLine>*, std::vector<verible::VectorTree<verible::UnwrappedLine>, std::allocator<verible::VectorTree<verible::UnwrappedLine> > > > > const&, verilog::formatter::FormatStyle const&), verible::container_iterator_range<__gnu_cxx::__normal_iterator<verible::VectorTree<verible::UnwrappedLine>*, std::vector<verible::VectorTree<verible::UnwrappedLine>, std::allocator<verible::VectorTree<verible::UnwrappedLine> > > > > const&, verilog::formatter::FormatStyle const&> (
    __f=@0x555555914448: 0x55555559dab2 <verilog::formatter::AlignActualNamedPorts(verible::TokenPartitionRange const&, verilog::formatter::FormatStyle const&)>)
    at /usr/include/c++/10/bits/invoke.h:60
#10 0x00005555555ac338 in std::__invoke_r<std::vector<verible::AlignablePartitionGroup, std::allocator<verible::AlignablePartitionGroup> >, std::vector<verible::AlignablePartitionGroup, std::allocator<verible::AlignablePartitionGroup> > (*&)(verible::container_iterator_range<__gnu_cxx::__normal_iterator<verible::VectorTree<verible::UnwrappedLine>*, std::vector<verible::VectorTree<verible::UnwrappedLine>, std::allocator<verible::VectorTree<verible::UnwrappedLine> > > > > const&, verilog::formatter::FormatStyle const&), verible::container_iterator_range<__gnu_cxx::__normal_iterator<verible::VectorTree<verible::UnwrappedLine>*, std::vector<verible::VectorTree<verible::UnwrappedLine>, std::allocator<verible::VectorTree<verible::UnwrappedLine> > > > > const&, verilog::formatter::FormatStyle const&> (
    __fn=@0x555555914448: 0x55555559dab2 <verilog::formatter::AlignActualNamedPorts(verible::TokenPartitionRange const&, verilog::formatter::FormatStyle const&)>)
    at /usr/include/c++/10/bits/invoke.h:115
[... cut ...]
#26 0x0000555555570e96 in formatOneFile (filename=..., lines_to_format=...) at verilog/tools/formatter/verilog_format.cc:260
#27 0x0000555555571704 in main (argc=2, argv=0x7fffffffdc68) at verilog/tools/formatter/verilog_format.cc:347
```
</details>

It looks like the [uwline.Origin() in verilog/formatting/align.cc](https://github.com/google/verible/blob/4c451f7899773a3fa94b3e93605fa54fc277ce99/verilog/formatting/align.cc#L148) is `nullptr`, causing the segmentation fault in the call to `SymbolCastToNode()`.

kbrunham-intel, 2026-09-22:
@hzeller I retested the reduced case from this issue with `verible-verilog-format` built from tag `v0.0-4294-gc1d8f5e8`. I could not reproduce the problem (formatter exits 0; output keeps the expected layout / converges). I would suggest closing this issue.

Instantiation with a comment between `.clk()` and the following comma formats (exit 0) instead of failing with no output.