chipsalliance/verible #778: Add verible to buildkite

mithro, 2021-04-21:
**Summary**

Page at https://github.com/bazelbuild/continuous-integration/blob/master/buildkite/README.md says;

> - _**Want to test your project on Bazel CI? Simply file a request via [this link](https://github.com/bazelbuild/continuous-integration/issues/new?template=adding-your-project-to-bazel-ci.md&title=Request+to+add+new+project+[PROJECT_NAME]&labels=new-project)!**_

We should probably do that?

mithro, 2021-04-21:
https://chromium.googlesource.com/external/github.com/google/flatbuffers/+/refs/heads/master/.bazelci/presubmit.yml?autodive=0

mithro, 2021-04-21:
https://github.com/grpc/grpc/blob/master/.bazelci/presubmit.yml
```yaml

# Configuration file for Bazel CI [1].
#
# Also testing on Bazel CI in addition of our normal CI workflow
# ensures that gRPC is tested against Bazel@HEAD and stays compatible
# with the latest release.
#
# See [2,3] in case you have questions.
#
# [1] https://github.com/bazelbuild/continuous-integration
# [2] https://github.com/grpc/grpc/issues/19171
# [3] https://github.com/grpc/grpc/pull/20784
---
# TODO(yannic): Ideally, we should also enable buildifier and all platforms should test `//...`.
platforms:
  ubuntu1604:
    build_targets:
      - //:all
      - //src/proto/...
      - //src/python/...
    test_targets:
      - //:all
      - //src/proto/...
      - //src/python/...
```

hzeller, 2021-04-21:
What advantages would it give us ?
It sounds it would make it harder to do things such as releases or run on different platforms and would make things more brittle as we would rely on some additional services but without any benefits that would be valuable for us.