chipsalliance/verible #806: Feature Suggestion: PyPi package

mballance, 2021-05-01:
Hello,
    I'm interested in using Verible via the Python interface in another tool. I'd like to have pre-built Verible packages available on PyPi that embed the appropriate executables and Python code, since this simplifies dependencies. Is this something that fits within the scope of the Verible project? 

Best Regards,
Matthew

mithro, 2021-05-01:
If Verible has a Python API, having a PyPi package seems logical.

mballance, 2021-05-01:
@mithro, agreed of course. Verible provides some Python infrastructure that provides a Python object API based on reading the JSON output by the syntax parser. I think the key question is whether this Python API is considered a primary part of the Verible tool, or just an example. If it's a primary part of the Verible tool, then having Verible release a PyPi package makes sense. If it's just an example (ie not a primary part of Verible), then perhaps a separate project to develop and maintain the Python infrastructure that reads Verible JSON makes more sense.

mithro, 2021-05-01:
I don't see any reason to exclude the Python API but reserve the right to change my opinion in the future given new information.