chipsalliance/verible #812: verible-verilog-format unable to preprocess macros correctly

tianrui-wei, 2021-05-09:
**Test case**

```systemverilog
`define width(x) [x-1:0]
module counter (
	input clk,    // Clock
	input rst_n,  // Asynchronous reset active low
	output `width(9) value
);

	logic [8:0] value_q, value_d;

	always_ff @(posedge clk) begin : proc_value_r
		if(~rst_n) begin
			value_q <= '0;
		end else begin
			value_q <= value_d;
		end
	end

	always_comb begin : proc_value_n
		value_d = value_q + 9'h2;
	end

    assign value = value_q;
	

endmodule
```

Include any options or configuration used.

no option or configuration

**Actual output**

```systemverilog
// This doesn't look right.
```

Include any possible diagnostic messages from the formatter.

counter.sv: counter.sv:5:9: syntax error, rejected "`width" (syntax-error).
counter.sv:10:2: syntax error, rejected "always_ff" (syntax-error).
counter.sv:13:3: syntax error, rejected "end" (syntax-error).
counter.sv:15:3: syntax error, rejected "end" (syntax-error).
counter.sv:20:2: syntax error, rejected "end" (syntax-error).
counter.sv:25:1: syntax error, rejected "endmodule" (syntax-error).

**Expected or suggested output**

There should not be errors from the formatter

```systemverilog
`define width(x) [x-1:0]
module counter (
	input clk,    // Clock
	input rst_n,  // Asynchronous reset active low
	output `width(9) value
);

	logic [8:0] value_q, value_d;

	always_ff @(posedge clk) begin : proc_value_r
		if(~rst_n) begin
			value_q <= '0;
		end else begin
			value_q <= value_d;
		end
	end

	always_comb begin : proc_value_n
		value_d = value_q + 9'h2;
	end

    assign value = value_q;
	

endmodule
```

tgorochowik, 2021-05-20:
@fangism @hzeller I guess this will have to wait for a proper preprocessor?

fangism, 2021-05-20:
> @fangism @hzeller I guess this will have to wait for a proper preprocessor?

Known limitation:
The current parser attempts to ingest _unpreprocessed_ source code, and only supports preprocessor macros in limited places.  Formatting is an inherently unpreprocessed operation.  A proper preprocessor is needed when actually evaluating semantics.  

To handle preprocessing macros more robustly, we'd have to use a different parser (same lexer) for the purposes of formatting (similar in strategy to clang-format).  The implementation goal of the formatting-parser would be to produce the same token-partition-tree structure that is the current one (language-agnostic), and leverage most of the existing formatting code.

fangism, 2021-05-20:
> Known limitation:
> The current parser attempts to ingest _unpreprocessed_ source code, and only supports preprocessor macros in limited places. Formatting is an inherently unpreprocessed operation. A [...]

Another possible strategy might be to attempt to recover from syntax errors and use text-preserving placeholders in the syntax-tree, but this is subject to the limits of error-recovery.

Since the definition of the macro is available, it might be possible to do best-effort pseudo-preprocessing, whereby the syntax evaluation is done on the partially _expanded_ source code, but the formatting manipulations operate on the _un-expanded_ source code.  The limitation of this approach is that macro definitions aren't necessarily available in the file to be formatted, perhaps coming from an `` `include``.

snsokolov, 2022-02-10:
Just speaking from my experience. It's a long known issue https://github.com/chipsalliance/verible/labels/preprocessor and as David mentioned above "macro definitions aren't necessarily available in the file to be formatted".
Personally I'd recommend avoiding any macro substitutions which are not falling at syntactic construct boundaries and prefer macros that are complete syntactic construct (I.e. like `ASSERT() or `FF() or `uvm_info() macros, these are all fine).

fangism, 2022-02-11:
Looking at your original example, one change allows the formatter to work:

```systemverilog
define width(x) x-1:0
module counter (
	input clk,    // Clock
	input rst_n,  // Asynchronous reset active low
	output [`width(9)] value
);
...
```

In most places where an _expression_ is allowed, a macro call is also allowed by the parser today.