chipsalliance/verible #813: Always expand coverpoint statements

sriyerg, 2021-05-12:
**Test case**

```systemverilog
// Input to the formatter, preferably a reduced test case.
 cp_handshake_complete: coverpoint status {
    bins complete = {AlertAckComplete};
  }
```

Include any options or configuration used.

**Actual output**

```systemverilog
// This doesn't look right.
  cp_handshake_complete: coverpoint status {bins complete = {AlertAckComplete};}
```

Include any possible diagnostic messages from the formatter.

**Expected or suggested output**
Original is better.

Citations to published style guides would help.
Don't have one yet (I think), but will add soon.

snsokolov, 2021-05-12:
> Citations to published style guides would help.

Not sure about other style guides, this not regulated by the internal style guide.

It looks to me that formatter is trying to keep short statements like this on one line to make it more concise, but +1 that the original is looking more readable.

msfschaffner, 2021-05-12:
If we decide to always expand coverpoints, it probably makes sense to do the same for constraints as well for consistency (as indicated by @sriyerg here: #445).

fangism, 2021-06-09:
General drive-by comment:
When there is a request to expand a section, the first thing I ask is whether a well-placed `//comment` can force that expansion, without having to add a control switch -- clang-format respects comments to force expansion in many places.  Some of the formatter logic is intended to respect comments to do that.  I understand that some users don't want to add extra comments.

PR #851 (2021-06-21): Optionally expand coverpoints
Add a switch to the formatter to always expand coverpoint declarations, even if they would fit on one line.
The option is `--expand_coverpoints` and is disabled by default.

Fixes #813.
closed 2021-06-21 by tgorochowik