chipsalliance/verible #818: Can't install building from source

alessandroaimar, 2021-05-15:
After successfully building verible with bazel, I tried to install it using the command specified on the readme and got this error

/opt/eda/vscode/verible/git_repo/bazel run //:install -c opt -- /opt/eda/vscode/verible/bin 
INFO: Analyzed target //:install (0 packages loaded, 0 targets configured).
INFO: Found 1 target...
Target //:install up-to-date:
  bazel-bin/_install_gen.sh
  bazel-bin/install
INFO: Elapsed time: 0.560s, Critical Path: 0.11s
INFO: 1 process: 1 internal.
INFO: Build completed successfully, 1 total action
INFO: Build completed successfully, 1 total action
installer.bash ERROR: Unexpected option '--'
Usage: bazel run @com_google_verible//:install [-c opt] [--] [-s] [--] /INSTALL_PREFIX

hzeller, 2021-05-17:
Are you on MacOS ? It is known that the install script does not work there. I think you need to install the gnu `install` utility. Might be `brew install coreutils` ?
I think @johanvdhaegen can help.

alessandroaimar, 2021-05-17:
Centos 8.3

Note that we realized that the installation actually completes successfully and the tests pass, so this may be just a script error

mglb, 2022-03-18:
> ```
> installer.bash ERROR: Unexpected option '--'
> Usage: bazel run @com_google_verible//:install [-c opt] [--] [-s] [--] /INSTALL_PREFIX`
> ```
This means that the `install` (which wraps `install.bash`) script is executed instead of system's `install` tool. This could happen when `PATH` environment variable contains empty entry (e.g. `:/usr/local/bin:/usr/bin:...`; note the `:` on the beginning) or `.`. Such entry makes current working directory a viable path for searching executables.