chipsalliance/verible #824: Crash if class has double semicolon item

hoaluvn, 2021-05-19:
**Test case**

```systemverilog
class sample;
bit a;;
bit b;
endclass
```

Works for module, program but not class

**Error message output**
F20210519 17:24:37.693542 23043 logging.h:39] Check failed: t != nullptr unwrapped_line.Origin()
*** Check failure stack trace: ***
    @           0x4b59ef  google::LogMessage::SendToLog()
    @           0x4b4879  google::LogMessage::Flush()
    @           0x4b6249  google::LogMessageFatal::~LogMessageFatal()
    @           0x450950  verible::AlignablePartitionGroup::CalculateAlignmentSpacings()
    @           0x452a2e  verible::AlignablePartitionGroup::Align()
    @           0x436f73  verilog::formatter::TabularAlignTokenPartitions()
    @           0x423f15  verible::VectorTree<>::ApplyPreOrder()
    @           0x423f31  verible::VectorTree<>::ApplyPreOrder()
    @           0x429441  verilog::formatter::Formatter::Format()
    @           0x42af32  verilog::formatter::FormatVerilog()
    @           0x414829  main
    @     0x7fcf3ab12555  __libc_start_main
    @           0x419a9e  (unknown)
    @              (nil)  (unknown)

hzeller, 2021-05-19:
Something going on in the formatter. Lukasz, can you (or delegate to someone) to figure out root cause ?

ldalek-antmicro, 2021-05-19:
@hzeller Got quick-fix for that here: #825

PR #825 (2021-05-19): formatter: Fix crash caused by an alignment of null statement
Fixes #824
closed 2021-05-19 by hzeller