chipsalliance/verible #826: WindowsBuild: `verible-* --version` doesn't print version

mglb, 2021-05-19:
This happens at least in CI (example: https://github.com/google/verible/pull/823/checks?check_run_id=2620255630#step:4:658 ):
```
$ verible-verilog-syntax.exe --version
Built   2021-05-19T12:49:01Z
$
```

Output on linux for comparison:
```
$ verible-verilog-syntax --version
v0.0-1217-gb5506e2
Commit  2021-05-13
Built   2021-05-13T14:55:29Z
$
```
I guess https://github.com/google/verible/blob/master/bazel/build-version.sh fails to get info from git.

hzeller, 2021-05-19:
I wonder maybe it is as simple has needing to install `git` as dependency in the windows-ci ? But of course could also be that just the shell-scripting is not working the same.

@corco might be able to spot what is going on.

corco, 2021-05-19:
Git is already installed on CI, I will check that on my end.

PR #827 (2021-05-20): Replaced build-version.sh with build-version.py
Fixes #826

The bazel option `--workspace_status_command` must point to an executable file, and on Windows bash shell script are not executable. Fortunately, Python scripts are executable and I opted to replace the current bash script with a Python one.

The alternative would be to have separate scripts for Linux/MacOS/Windows.
closed 2021-05-20 by hzeller