chipsalliance/verible #831: formatter output is lexically different from input

somyadashora, 2021-05-23:
**Test case**

```systemverilog
myfile.sv: Formatted output is lexically different from the input.    Please file a bug.  Details:
Mismatch in token sequence lengths: 1307 vs. 1306
First mismatched token [482]: ("// end of line comment") (#748: "// Not a Leaf Entry") vs. ("// end of line comment") (#748: "// Not a Leaf Entry            else")
; problematic formatter output is
```

Include any options or configuration used.

**Actual output**

```systemverilog
// This doesn't look right.
```

Include any possible diagnostic messages from the formatter.

**Expected or suggested output**

```systemverilog
// This result would look better from the formatter.
```

Citations to published style guides would help.

hzeller, 2022-01-21:
You didn't include an example. From the error message I've constructed something that could result in
```SystemVerilog
module foo();
  always begin
    if (a)
      b <= 1;
    // comment
    else begin
      b <= 2;
    end
  end
endmodule
```

Output
```
Mismatch in token sequence lengths: 26 vs. 25
First mismatched token [15]: ("// end of line comment") (#748: "// comment") vs. ("// end of line comment") (#748: "// comment    else")
; problematic formatter output is
```

.. where it mangles the else into the comment:

```SystemVerilog
module foo ();
  always begin
    if (a) b <= 1;
    // comment    else
    begin
      b <= 2;
    end
  end
endmodule
```

hzeller, 2022-01-21:
Does this have the same root-cause as #1113 @mglb ?

mglb, 2022-01-21:
Not really. In this one there's assumption that the first subpartition is an `else` keyword, but actually it is a comment.
In #1113 there's assumption that trailing `;` can be always safely merged with a partition preceding it.

PR #1169 (2022-02-02): Prevent merge of EOL comment and regular code in else-if/else-begin
Fixes #831
closed 2022-02-02 by hzeller