chipsalliance/verible #833: Syntax error, rejected inf

treforsouthwell, 2021-05-24:
The parser is failing when a logic is called 'inf':

test2.sv:5:13: syntax error, rejected "inf" (syntax-error).

Short summary.

verible-verilog-lint test2.sv

```systemverilog

module fred;

logic fred;
logic joe, pete;
logic here, inf;

endmodule

```

Expected to parse normally as inf is not a reserved word in SV
Cadence xcelium and synopsys VCS compiles this code fine

Khadija411, 2022-04-11:
**verilog_lexer_unittest.cc:1166** 
[image]

_inf_ is listed in the C++ standard library as a keyword

**verilog_lexer_unittest.cc:865** `static std::initializer_list<LexerTestData> kKeywordTests`

In addition to the keywords within Verilog HDL _inf_ is also included.
The keyword _inf_ may be used to indicate infinity. If preceded by a negative sign, it
indicates negative infinity. For example,
`parameter real val3=0 from [0:inf) exclude (10:20) exclude (30:40];`

mglb, 2022-04-12:
Bug confirmed - `inf` is NOT reserved in SystemVerilog.

It is a thing in Verilog A/AMS, which is not supported by Verible.