chipsalliance/verible #838: Add configuration option for alignment of dimensions

wtatarski, 2021-05-28:
In #433 we highlighted 2 options for alignment of dimensions in function argument list:
```
  import "DPI-C" context function void c_dpi_aes_crypt_message(
    input  bit              impl_i,
    input  bit              op_i,
    input  bit        [5:0] mode_i,
    input  bit  [3:0][31:0] iv_i,
    input  bit        [2:0] key_len_i,
    input  bit  [7:0][31:0] key_i,
    input  bit        [7:0] data_i[],
    output bit        [7:0] data_o[]
  );
```
Or:
```
  import "DPI-C" context function void c_dpi_aes_crypt_message(
    input  bit             impl_i,
    input  bit             op_i,
    input  bit [5:0]       mode_i,
    input  bit [3:0][31:0] iv_i,
    input  bit [2:0]       key_len_i,
    input  bit [7:0][31:0] key_i,
    input  bit [7:0]       data_i   [],
    output bit [7:0]       data_o   []
  );
```

A switch should be added for deciding whether formatter should flush dimensions right or left.

tgorochowik, 2021-05-28:
from @hzeller :
> I'd probably tie that together with whatever happens in module ports for consistency.

mglb, 2021-07-29:
Since each dimension creates its own column, I guess this should work like this:

left aligned:

```systemverilog
import "DPI-C" context function void c_dpi_aes_crypt_message(
  input  bit                  impl_i,
  input  bit                  op_i,
  input  bit [ 5:0]           mode_i,
  input  bit [ 3:0][ 4:0 ]    iv_i,
  input  bit [12:0]           key_len_i,
  input  bit [ 7:0][31:24][2] key_i,
  input  bit [ 7:0]           data_i   [],
  output bit [ 7:0]           data_o   []
);
```

right aligned:

```systemverilog
import "DPI-C" context function void c_dpi_aes_crypt_message(
  input  bit                    impl_i,
  input  bit                    op_i,
  input  bit             [ 5:0] mode_i,
  input  bit      [ 3:0 ][ 4:0] iv_i,
  input  bit             [12:0] key_len_i,
  input  bit [7:0][31:24][   2] key_i,
  input  bit             [ 7:0] data_i   [],
  output bit             [ 7:0] data_o   []
);
```

closed 2021-09-20 by hzeller