chipsalliance/verible #853: Crash when enum constant comma separation in new line after EOL comment.

hzeller, 2021-06-10:
**Test case**

If the comma that separates enumeration constants _after_ a one-line comment, then the formatter will crash.

```systemverilog
typedef enum {
   a // foo
  ,b } c;
```

```
bazel build -c dbg verilog/tools/formatter:verible-verilog-format
bazel-bin/verilog/tools/formatter/verible-verilog-format /tmp/a.sv
F20210610 14:41:15.854058 24849 align.cc:252] Check failed: pos_iter != cend 
*** Check failure stack trace: ***
    @     0x5585b590d504  google::LogMessage::Fail()
    @     0x5585b590d44f  google::LogMessage::SendToLog()
    @     0x5585b590cd76  google::LogMessage::Flush()
    @     0x5585b591055c  google::LogMessageFatal::~LogMessageFatal()
    @     0x5585b581ad74  verible::FillAlignmentRow()
    @     0x5585b581d5e0  verible::AlignablePartitionGroup::CalculateAlignmentSpacings()
    @     0x5585b581ef01  verible::AlignablePartitionGroup::Align()
    @     0x5585b57e4c42  verilog::formatter::TabularAlignTokenPartitions()
    @     0x5585b57d5e01  _ZZN7verilog9formatter9Formatter6FormatERKNS0_16ExecutionControlEENKUlRN7verible10VectorTreeINS5_13UnwrappedLineEEEE_clES9_
    @     0x5585b57d71a1  _ZSt13__invoke_implIvRZN7verilog9formatter9Formatter6FormatERKNS1_16ExecutionControlEEUlRN7verible10VectorTreeINS6_13UnwrappedLineEEEE_JSA_EET_St14__invoke_otherOT0_DpOT1_
    @     0x5585b57d6ece  _ZSt10__invoke_rIvRZN7verilog9formatter9Formatter6FormatERKNS1_16ExecutionControlEEUlRN7verible10VectorTreeINS6_13UnwrappedLineEEEE_JSA_EENSt9enable_ifIX16is_invocable_r_vIT_T0_DpT1_EESE_E4typeEOSF_DpOSG_
    @     0x5585b57d6bc8  _ZNSt17_Function_handlerIFvRN7verible10VectorTreeINS0_13UnwrappedLineEEEEZN7verilog9formatter9Formatter6FormatERKNS7_16ExecutionControlEEUlS4_E_E9_M_invokeERKSt9_Any_dataS4_
    @     0x5585b57d97cd  std::function<>::operator()()
    @     0x5585b57d811f  verible::VectorTree<>::ApplyPreOrder()
    @     0x5585b57d8189  verible::VectorTree<>::ApplyPreOrder()
    @     0x5585b57d8189  verible::VectorTree<>::ApplyPreOrder()
    @     0x5585b57d7b8d  verible::TreeUnwrapper::ApplyPreOrder()
    @     0x5585b57d616d  verilog::formatter::Formatter::Format()
    @     0x5585b57d4194  verilog::formatter::FormatVerilog()
    @     0x5585b57b58dd  formatOneFile()
    @     0x5585b57b614b  main
    @     0x7f2f02228d0a  __libc_start_main
    @     0x5585b57b508a  _start
    @              (nil)  (unknown)
```

a non-crashing formatter should leave such constructs alone possibly and leave the comma out of place. Ideally could figure out that the comma and the comment can be swapped to get the following: 
```systemverilog
typedef enum {
   a,  // foo
   b } c;
```

hzeller, 2021-06-10:
b/190735211

hzeller, 2021-09-14:
@mglb with your familiarity with the formatter now, could you have a look ?

First step is to _not_ make it crash Also possibly similar reason in #920), and possibly leave the formatting untouched when we can't deal with it.

Then, longer term consider the prefix-comma and the consequence that it is common to have EOL comments between field and comma as one style-option, which should probably be formulated as separate formatter feature request.. Maybe using some autodetecting what style is used, and then run with it ?

mglb, 2021-09-14:
This doesn't seem to crash with current master, but the result is even worse:
```
typedef enum {
  a  // foo,
  b
} c;
```

Will look at it.

commit 93473bb8 (2021-10-19): Do not merge ',' with partitions containing EOL comment

Fixes #853
closed 2021-10-19 by hzeller