chipsalliance/verible #862: Grammar railroad diagram

mingodad, 2021-07-06:
I've done a experimental tool to convert bison grammars to a kind of EBNF understood by https://www.bottlecaps.de/rr/ui to generate railroad diagrams see bellow the converted `verilog/parser/verilog.y` and with some hand made changes to allow view it at https://www.bottlecaps.de/rr/ui the order of the rules could be changed to a better view of the railroad diagrams. Copy and paste the EBNF bellow on https://www.bottlecaps.de/rr/ui tab Edit Grammar then switch to the tab View Diagram.

```
/*
From https://raw.githubusercontent.com/google/verible/master/verilog/parser/verilog.y
*/

source_text ::=
	description_list
	| /* empty */

GenericIdentifier ::=
	SymbolIdentifier
	| EscapedIdentifier
	| MacroIdentifier
	| KeywordIdentifier
	| unary_builtin_function
	| binary_builtin_function

KeywordIdentifier ::=
	TK_branch
	| TK_access
	| TK_exclude
	| TK_flow
	| TK_from
	| TK_ground
	| TK_connect
	| TK_ddx
	| TK_ddt
	| TK_idt
	| TK_idtmod
	| TK_split
	| TK_merged
	| TK_timer
	| TK_above
	| TK_discrete
	| TK_initial_step
	| TK_final_step
	| TK_sample
	| TK_bool

preprocessor_directive ::=
	preprocessor_control_flow
	| preprocessor_action

preprocessor_if_header ::=
	PP_ifdef PP_Identifier
	| PP_ifndef PP_Identifier

preprocessor_elsif_header ::=
	PP_elsif PP_Identifier

preprocessor_control_flow ::=
	preprocessor_if_header
	| preprocessor_elsif_header
	| PP_else
	| PP_endif

preprocessor_action ::=
	PP_undef PP_Identifier
	| PP_include preprocess_include_argument
	| PP_define PP_Identifier PP_define_body
	| PP_define PP_Identifier '(' macro_formals_list_opt ')' PP_define_body

macro_formals_list_opt ::=
	macro_formals_list
	| /* empty */

macro_formals_list ::=
	macro_formals_list ',' macro_formal_parameter
[... 3974 lines cut ...]
TK_OR_EQ ::= "|="
TK_XOR_EQ ::= "^="
TK_LS_EQ ::= "<<="
TK_RS_EQ ::= ">>="
TK_LS_EQ ::= "<<<="
TK_RSS_EQ ::= ">>>="
TK_INCR ::= "++"
TK_DECR ::= "--"
TK_LP ::= "'{"
TK_SCOPE_RES ::= "::"
TK_COLON_EQ ::= ":="
//yytext[0] ::= "://"
//yytext[0] ::= ":/*"
  /* Prevent ":/" from consuming the start of a comment. */
TK_COLON_DIV ::= ":/"
TK_POUNDMINUSPOUND ::= "#-#"
TK_POUNDEQPOUND ::= "#=#"
TK_POUNDPOUND ::= "##"
TK_LBSTARRB ::= "[*]"
TK_LBPLUSRB ::= "[+]"
TK_LBSTAR ::= "[*"
TK_LBEQ ::= "[="
TK_LBRARROW ::= "[->"
TK_ATAT ::= "@@"
```

hzeller, 2022-08-26:
This is neat!
You mention a tool you wrote to do the conversion - is it available somewhere so that we could maybe integrate this in the build-process to auto-generate documentation ?

(sorry for the delay, just seeing this :) )

mingodad, 2022-08-26:
There is this online tool https://www.bottlecaps.de/convert/ form the same author of the railroad diagram tool.

mingodad, 2022-08-26:
Also I extended bison/byacc/lemon here https://github.com/mingodad/lalr-parser-test to also generate an EBNF compatible with https://www.bottlecaps.de/rr/ui