chipsalliance/verible #878: Formatting bug on unique-case statement

msfschaffner, 2021-08-04:
**Test case (test.sv below)**
```systemverilog
  always_comb begin : p_mode
    aon_wkup_pulse_o = 1'b0;
    cnt_en     = 1'b0;
    if (wkup_en_i) begin
      unique case (wkup_mode_i)
        Negedge:   aon_wkup_pulse_o = falling;
        Edge:      aon_wkup_pulse_o = rising | falling;
        HighTimed: begin
          cnt_en = filter_out_d;
          aon_wkup_pulse_o = cnt_eq_th;
        end
        LowTimed: begin
          cnt_en = ~filter_out_d;
          aon_wkup_pulse_o = cnt_eq_th;
        end
        // Default to rising
        default: aon_wkup_pulse_o = rising;
      endcase
    end
  end

```

Run using the `util/verible-format.py` script from the OpenTitan repository:
```
util/verible-format.py --inplace -v -f test.sv 
WARNING: Operating in-place - so make sure to make a backup or run this on an experimental branch
Using Verible formatter version: v0.0-1213-g9e5c085
repo_top: /usr/local/google/home/msf/Projects/opentitan
verible exec: /tools/verible/v0.0-1213-g9e5c085/bin/verible-verilog-format
verible args: ['--formal_parameters_indentation=indent', '--named_parameter_indentation=indent', '--named_port_indentation=indent', '--port_declarations_indentation=indent']
files to format: ['/usr/local/google/home/msf/Projects/opentitan/test.sv']
ERROR: /usr/local/google/home/msf/Projects/opentitan/test.sv
/usr/local/google/home/msf/Projects/opentitan/test.sv: Re-formatted text does not match formatted text; formatting failed to converge!  Please file a bug.
Original: --lines: 
  always_comb begin : p_mode
    aon_wkup_pulse_o = 1'b0;
    cnt_en     = 1'b0;
    if (wkup_en_i) begin
      unique case (wkup_mode_i)
        Negedge:   aon_wkup_pulse_o = falling;
        Edge:      aon_wkup_pulse_o = rising | falling;
        HighTimed: begin
          cnt_en = filter_out_d;
          aon_wkup_pulse_o = cnt_eq_th;
        end
        LowTimed: begin
          cnt_en = ~filter_out_d;
          aon_wkup_pulse_o = cnt_eq_th;
        end
        // Default to rising
        default: aon_wkup_pulse_o = rising;
      endcase
    end
  end
Formatted:
always_comb begin : p_mode
  aon_wkup_pulse_o = 1'b0;
  cnt_en           = 1'b0;
  if (wkup_en_i) begin
    unique case (wkup_mode_i)
      Negedge: aon_wkup_pulse_o = falling;
      Edge:    aon_wkup_pulse_o = rising | falling;
      HighTimed: begin
        cnt_en = filter_out_d;
        aon_wkup_pulse_o = cnt_eq_th;
      end
      LowTimed: begin
        cnt_en = ~filter_out_d;
        aon_wkup_pulse_o = cnt_eq_th;
      end
      // Default to rising
      default: aon_wkup_pulse_o = rising;
    endcase
  end
end
Re-formatted:
always_comb begin : p_mode
  aon_wkup_pulse_o = 1'b0;
  cnt_en           = 1'b0;
  if (wkup_en_i) begin
    unique case (wkup_mode_i)
      Negedge: aon_wkup_pulse_o = falling;
      Edge:    aon_wkup_pulse_o = rising | falling;
      HighTimed: begin
cnt_en = filter_out_d;
aon_wkup_pulse_o = cnt_eq_th;
end
      LowTimed: begin
cnt_en = ~filter_out_d;
aon_wkup_pulse_o = cnt_eq_th;
end
      // Default to rising
      default: aon_wkup_pulse_o = rising;
    endcase
  end
end
Diffs are:
 always_comb begin : p_mode
   aon_wkup_pulse_o = 1'b0;
   cnt_en           = 1'b0;
   if (wkup_en_i) begin
     unique case (wkup_mode_i)
       Negedge: aon_wkup_pulse_o = falling;
       Edge:    aon_wkup_pulse_o = rising | falling;
       HighTimed: begin
-        cnt_en = filter_out_d;
-        aon_wkup_pulse_o = cnt_eq_th;
-      end
+cnt_en = filter_out_d;
+aon_wkup_pulse_o = cnt_eq_th;
+end
       LowTimed: begin
-        cnt_en = ~filter_out_d;
-        aon_wkup_pulse_o = cnt_eq_th;
-      end
+cnt_en = ~filter_out_d;
+aon_wkup_pulse_o = cnt_eq_th;
+end
       // Default to rising
       default: aon_wkup_pulse_o = rising;
     endcase
   end
 end
; problematic formatter output is
always_comb begin : p_mode
  aon_wkup_pulse_o = 1'b0;
  cnt_en           = 1'b0;
  if (wkup_en_i) begin
    unique case (wkup_mode_i)
      Negedge: aon_wkup_pulse_o = falling;
      Edge:    aon_wkup_pulse_o = rising | falling;
      HighTimed: begin
        cnt_en = filter_out_d;
        aon_wkup_pulse_o = cnt_eq_th;
      end
      LowTimed: begin
        cnt_en = ~filter_out_d;
        aon_wkup_pulse_o = cnt_eq_th;
      end
      // Default to rising
      default: aon_wkup_pulse_o = rising;
    endcase
  end
end
<<EOF>>
```

msfschaffner, 2021-08-04:
The error disappears when all cases use `begin` `else` blocks:
```systemverilog
  always_comb begin : p_mode
    aon_wkup_pulse_o = 1'b0;
    cnt_en           = 1'b0;
    if (wkup_en_i) begin
      unique case (wkup_mode_i)
        Negedge: begin
          aon_wkup_pulse_o = falling;
        end
        Edge: begin
          aon_wkup_pulse_o = rising | falling;
        end
        HighTimed: begin
          cnt_en = filter_out_d;
          aon_wkup_pulse_o = cnt_eq_th;
        end
        LowTimed: begin
          cnt_en = ~filter_out_d;
          aon_wkup_pulse_o = cnt_eq_th;
        end
        // Default to rising
        default: begin
          aon_wkup_pulse_o = rising;
        end
      endcase
    end
  end

```

hankhsu1996, 2023-04-16:
Add another example to cause the "Re-formatted text does not match formatted text" error. 

```verilog
module Test;
  initial begin
// verilog_format: off
    case (1)
      0,
      1: $display("0 or 1");
      default: $display("default");
    endcase
// verilog_format: on
  end
endmodule
```