chipsalliance/verible #885: verible-verilog-syntax --export_json emits nothing

dworvos, 2021-08-13:
Hi,

I am using the version v0.0-1320-gf88614c on CentOS 7 and I'm having trouble getting the --export_json flag to work.

Most likely I'm using the tool wrong but I created a simple verilog file `foo.sv` as provided in the documentation:

```
// This is module foo.
module foo(input a, b, output z);
endmodule : foo
```

However, when I run verible-verilog-syntax with --export_json it outputs this:

```
{
        "foo.sv": null
}
```

However, using the other flags such as  --printtokens and --printtree gives reasonable output. Here's my full command line for reference:

```
verible-verilog-syntax --lang sv --export_json foo.sv
```

Thanks in advance for your help.

hzeller, 2021-08-13:
@dworvos You also need to supply one of the `--printtree`, `--printtokens` or `--printrawtokens` option which tells `verible-verilog-syntax` what to output; the `--export_json` then tells it to do it in json format.

@mglb : Maybe if `--export_json` is given but none of the other output options is chosen, we should  print an info message what is missing ?
Testing: There is a python test in the example directory, but for direct testing of the json functionality (and documenting its execution) a test in  [verilog/tools/syntax/verilog_syntax_test.sh](https://github.com/chipsalliance/verible/blob/master/verilog/tools/syntax/verilog_syntax_test.sh) (or separate *.sh test) would probably also a good idea.

dworvos, 2021-08-13:
Ah thank you, that worked! I thought I tried with --printtree but turns out I invoked it with stdin which gave me null (and thus making me think I was missing something very fundamental):

`verible-verilog-syntax --export_json --lang sv --printtokens < foo.sv`

Turns out if I omit the stdin from the shell it works as expected, i.e.

`verible-verilog-syntax --export_json --lang sv --printtokens foo.sv`

Thanks again for your help!

mglb, 2021-08-23:
@hzeller

> @mglb : Maybe if `--export_json` is given but none of the other output options is chosen, we should print an info message what is missing ?

`verible-verilog-syntax` (with and without `--export_json`) without any `--print_*` switch is a valid use case - it checks for syntax errors. Examples:

* `verible-verilog-syntax xx.sv`:
  ```
  xx.sv:3:9: syntax error, rejected "foo" (syntax-error).
  ```

* `verible-verilog-syntax --export_json xx.sv`:
  ```
  {
          "xx.sv":
          {
                  "errors":
                  [
                          {
                                  "column": 8,
                                  "line": 2,
                                  "phase": "parse",
                                  "text": "foo"
                          }
                  ]
          }
  }
  ```

> Testing: There is a python test in the example directory, but for direct testing of the json functionality (and documenting its execution) a test in [...]

Good idea. I've created the issue for this: https://github.com/chipsalliance/verible/issues/891