chipsalliance/verible #887: Formatting of ternaries

msfschaffner, 2021-08-17:
**Test case**

```systemverilog
assign myvar = check                ? {10'b0, foo}       :
               (bar && (baz == '0)) ? hello_longlonglong :
               world_longlonglong   ? temp1_longlonglong : temp2_longlonglong;
```
Invoked using
```shell
util/verible-format.py --inplace --files
Using Verible formatter version: v0.0-1213-g9e5c085
```
in the OpenTitan repo (using the OpenTitan formatter configuration).

**Actual output**

The `{}` brackets seem to enforce a rather unconventional formatting for this ternary here:

```systemverilog
assign myvar = check ? {
  10'b0, foo
} : (bar && (baz == '0)) ? hello_longlonglong :
    world_longlonglong ? temp1_longlonglong : temp2_longlonglong;
```

A formatting like the original input above would be preferrable.

hzeller, 2021-08-17:
Also confirmed with current version `v0.0-1320-gf88614c`

```
verible-verilog-format - <<EOF
assign myvar = check                ? {10'b0, foo}       :
               (bar && (baz == '0)) ? hello_longlonglong :
               world_longlonglong   ? temp1_longlonglong : temp2_longlonglong;
EOF
```

``` SystemVerilog
assign myvar = check ? {
  10'b0, foo
} : (bar && (baz == '0)) ? hello_longlonglong :
    world_longlonglong ? temp1_longlonglong : temp2_longlonglong;
```

msfschaffner, 2021-09-01:
More examples for this:

before:
```systemverilog
  assign {dc_wrap, dc_htbt_d} = !(htbt_ctr_q == blink_param_x_i) ? {1'b0, dc_htbt_q} :
                                ((dc_htbt_q == duty_cycle_a_i) && pattern_repeat) ?
                                {1'b0, duty_cycle_a_i} : (htbt_direction) ?
                                {1'b0, dc_htbt_q} - {1'b0, blink_param_y_i} - 1'b1 :
                                {1'b0, dc_htbt_q} + {1'b0, blink_param_y_i} + 1'b1;
```
after:
```systemverilog
  assign {dc_wrap, dc_htbt_d} = !(htbt_ctr_q == blink_param_x_i) ? {
    1'b0, dc_htbt_q
  } : ((dc_htbt_q == duty_cycle_a_i) && pattern_repeat) ? {
    1'b0, duty_cycle_a_i
  } : (htbt_direction) ? {
    1'b0, dc_htbt_q
  } - {
    1'b0, blink_param_y_i
  } - 1'b1 : {
    1'b0, dc_htbt_q
  } + {
    1'b0, blink_param_y_i
  } + 1'b1;
```

before:
```systemverilog
  assign {phase_wrap, off_phase} = {1'b0, phase_delay_scaled[15:0]} +
                                   {1'b0, duty_cycle_scaled[15:0]};
```
after:
```systemverilog
  assign {phase_wrap, off_phase} = {
    1'b0, phase_delay_scaled[15:0]
  } + {
    1'b0, duty_cycle_scaled[15:0]
  };
```

the original formatting is probably not ideal either, but the `{}` bracket wrapping onto multiple lines really makes this hard to read...

closed 2021-09-23 by hzeller