chipsalliance/verible #899: syntax: `randomize` function named parameter erroneously rejected.

matlupi, 2021-08-31:
**Describe the bug**

If a class contains an object with a method with the `randomize` word used as a named parameter, this is reported as a syntax error.

**To Reproduce**

With `verible_example.sv` as 

```systemverilog
class my_env extends uvm_env;
  `uvm_component_utils(my_env)
  uvc_env      env;
  uvc_env_cfg  env_cfg;

  function new(string name="my_env", uvm_component parent=null);
    super.new(name, parent);
  endfunction

  function void build_phase(uvm_phase phase);
    super.build_phase(phase);
    env_cfg = uvc_env_cfg::type_id::create("env_cfg");
    env_cfg.add_active_agent(.randomize(1'b0),.base_unit(1ps));
  endfunction
endclass
```

Run 

```
verible-verilog-syntax verible_example.sv
```

**Actual behavior:**

Syntax error outputs (rejected valid syntax)

```
verible_example.sv:14:31: syntax error, rejected "randomize" (syntax-error).
```

**Expected behavior**

The syntax should be accepted.
It is not a SystemVerilog keyword according to [SystemVerilog-2017 Standard (LRM)](https://ieeexplore.ieee.org/document/8299595) Table B.1.
Also not present in reserved keywords in other tables (21-11, 22-2, 22-3, 22-4, 22-5, 22-6).

hzeller, 2021-08-31:
Yes, right now this is indeed used as separate keyword https://github.com/chipsalliance/verible/blob/master/verilog/parser/verilog.lex#L636

@fangism might have more context, but I think this is used here to cover the standard SystemVerilog functions for randomization, but randomize() in itself should only be special when called on an object AFAIK. @fangism do you know if there might be some other reason why this is treated as a keyword here, maybe some particular dialect you had in mind when working on the grammar ?

hzeller, 2021-08-31:
Also, we should maybe add this as a test-case in [sv-tests](https://github.com/chipsalliance/sv-tests).

fangism, 2021-08-31:
It is not a formal keyword, but it has a special form of trailing with-constraints (18.7) which requires it to be treated like a keyword in some contexts.  SystemVerilog has several sometimes-keywords like this, [example](https://github.com/chipsalliance/verible/blob/ee63d53999a4d2a0e506131b92c9e579507ae583/verilog/parser/verilog.lex#L446).  We could try to be a little smarter in the lexer to allow `randomize` to be used as a plain identifier in other contexts.

fangism, 2021-08-31:
If this cannot be solved in the lexer itself, the next best place is [LexicalContext](https://github.com/chipsalliance/verible/blob/ee63d53999a4d2a0e506131b92c9e579507ae583/verilog/parser/verilog_lexical_context.h#L227) which was designed to handle arbitrarily complex corner-cases with state-machine logic, which is a in-place pass over the lexical stream: tokens-in, tokens-out.  Here you can keep track of any token look-back, and alter the lexical stream, for example by changing token enumerations.