chipsalliance/verible #908: Bazel build error on Fedora 34

msagca, 2021-09-03:
I can successfully build Verible on Ubuntu machines but I'm having trouble on Fedora 34 (Workstation Edition). When I run `bazel build -c opt //...` I see the error message below.
```
ERROR: /home/mustafa/verible/verilog/parser/BUILD:122:1: @platforms//os:windows is not a valid configuration key for //verilog/parser:verilog_y_cc
ERROR: Analysis of target '//verilog/parser:verilog_y_cc' failed; build aborted: @platforms//os:windows is not a valid configuration key for //verilog/parser:verilog_y_cc
INFO: Elapsed time: 0.249s
INFO: 0 processes.
FAILED: Build did NOT complete successfully (0 packages loaded, 0 targets configured)
```
I don't know anything about the Bazel build system. How can I fix this issue?  
Thanks.

hzeller, 2021-09-03:
What is your bazel version ? I suspect it might not have the platform specific configuration yet.

 `bazel --version` should output something at least 4.0. Current stable is [4.2.1](https://github.com/bazelbuild/bazel/releases/tag/4.2.1), and there is even a [pre 5.0.0](https://github.com/bazelbuild/bazel/releases/tag/5.0.0-pre.20210826.1) out there, but haven't tested that yet.

To hack things up to make it work, you can just [remove the whole section](https://github.com/chipsalliance/verible/blob/master/verilog/parser/BUILD#L32-L37) in the file and try again.
```
 copts = select({
        "@platforms//os:windows": [],
        "//conditions:default": [
            "-Wno-implicit-fallthrough",
        ],
    }),
```

msagca, 2021-09-03:
Turns out my Bazel version was too old. I thought `dnf install bazel` would install the latest version but it did not. Now I'm getting another error but at least there is some progress. Thank you @hzeller.

closed 2021-09-03 by msagca