chipsalliance/verible #917: verible-verilog-project crashes on this sample file

hzeller, 2021-09-07:
Need to drill down what the problem is, but here some random file from ibex `examples/simple_system/rtl/ibex_simple_system.sv` make the project tool crash:

To reproduce
```
bazel build verilog/tools/project:verible-verilog-project

IBEX_DIR=/tmp/test/ibex
git clone https://github.com/lowrisc/ibex ${IBEX_DIR}

echo "${IBEX_DIR}/examples/simple_system/rtl/ibex_simple_system.sv" > /tmp/simple.list
bazel-bin/verilog/tools/project/verible-verilog-project symbol-table-defs --file_list_root=/ --file_list_path=/tmp/simple.list
```

Output
```
F20210907 16:05:04.105469 1043670 logging.h:39] Check failed: t != nullptr reference_branch_point_
*** Check failure stack trace: ***
    @     0x556d1ab1fd58  google::LogMessage::Fail()
    @     0x556d1ab1fca3  google::LogMessage::SendToLog()
    @     0x556d1ab1f5ca  google::LogMessage::Flush()
    @     0x556d1ab22dae  google::LogMessageFatal::~LogMessageFatal()
    @     0x556d1aa4d5e7  verible::DieIfNull<>()
    @     0x556d1aa4884e  verilog::SymbolTable::Builder::HandleIdentifier()
    @     0x556d1aa48c62  verilog::SymbolTable::Builder::Visit()
    @     0x556d1ab14250  verible::SyntaxTreeLeaf::Accept()
    @     0x556d1aaf5f81  verible::TreeContextVisitor::Visit()
    @     0x556d1aa46b7f  verilog::SymbolTable::Builder::Descend()
    @     0x556d1aa4699b  verilog::SymbolTable::Builder::Visit()
    ...
```

fangism, 2022-02-03:
Reduced test case:

```systemverilog
module foo;
  timer #(.N(1)) t;
endmodule
```

fangism, 2022-02-03:
The following code works:

```systemverilog
module foo;
  timer t(.x(y));
endmodule
```

So there's a hint.

They are supposed to create references to `timer::N` and `timer::x` in the same manner.  
I probably just forgot to do it for the parameter case when I handled the port case.

hzeller, 2022-02-24:
This works, if we don't use `timer` but any other identifier.

`timer`: What looks like a module name is [actually a keyword](https://github.com/chipsalliance/verible/blob/master/verilog/parser/verilog.y#L832)

.. and printing the syntax tree (with lower-case `timer` and a different spelling to not make it match that keyword shows that:

```
verible-verilog-syntax <(echo "module m; timer #(.N(1)) t; endmodule") --printtree | grep -i timer
                Leaf @0 (#"timer" @10-15: "timer")
```
```
verible-verilog-syntax <(echo "module m; tiMer #(.N(1)) t; endmodule") --printtree | grep -i timer
                Leaf @0 (#SymbolIdentifier @10-15: "tiMer")
```

It looks like the intent is to [treat is as identifier](https://github.com/chipsalliance/verible/blob/master/verilog/parser/verilog.y#L789), but that does currently not ripple through properly.

PR #1226 (2022-03-01): Improve KeywordIdentifier detection in symbol table using
IsIdentifierLike()

The overlapping keywords and identifiers are the result of
mis-classification in the SymbolTable by simply looking
at SymbolIdentifier; instead, use the IsIdentifierLike()
predicate function.

Also, while at it, reduce the number of special cases by
only keep the KeywordIdentifier tokens that are also used
in other places of the grammar.

Long term, we need to make sure to properly handle
KeywordIdentifiers in the whole stack; it seems like there
are other subtle bugs lurking.

One approach would be to change the tok [... cut]
commit 1a29ec77 (2022-03-01): Improve KeywordIdentifier detection in symbol table using
IsIdentifierLike()

The overlapping keywords and identifiers are the result of
mis-classification in the SymbolTable by simply looking
at SymbolIdentifier; instead, use the IsIdentifierLike()
predicate function.

Also, while at it, reduce the number of special cases by
only keep the KeywordIdentifier tokens that are also used
in other places of the grammar.

Long term, we need to make sure to properly handle
KeywordIdentifiers in the whole stack; it seems like there
are other subtle bugs lurking.

One approach would be to change the tok [... cut]
closed 2022-03-01 by hzeller