chipsalliance/verible #920: Segmentation fault on floating comma

zegervdv, 2021-09-08:
**Test case**

```systemverilog
module foo;

  bar#(
    .param_a(1)
    // pragma synthesis off
    ,
    .param_b(2)
    // pragma synthesis on
  ) m_bar ();

endmodule

```

Running formatting on the file gives a segmentation fault.
Linting on the module works fine.

When reshuffling the parameters in the list to avoid the comma on a single line it works fine:

``` systemverilog
module foo;

  bar#(
    // pragma synthesis off
    .param_b(2),
    // pragma synthesis on
    .param_a(1)
  ) m_bar ();

endmodule
```

It also works when the pragmas are not there.

So it seems to be a combination of the `// pragma ...` followed by a line with only a comma.

hzeller, 2021-09-09:
This looks similar to #853 , probably one fix will fix them both.

mglb, 2021-09-14:
This is not exactly the same as #853 . Stacktrace:

```c++
#0  0x00005555556dbea4 in verible::SymbolCastToNode (symbol=...) at common/text/tree_utils.cc:111
#1  0x00005555555a331f in verilog::formatter::IgnoreWithinActualNamedParameterPartitionGroup (partition=...) at verilog/formatting/align.cc:158
#2  0x00005555555b50c3 in std::_Function_handler<bool (verible::VectorTree<verible::UnwrappedLine> const&), bool (*)(verible::VectorTree<verible::UnwrappedLine> const&)>::_M_invoke(std::_Any_data const&, verible::VectorTree<verible::UnwrappedLine> const&) (__functor=..., __args#0=...) at /usr/include/c++/9/bits/std_function.h:285
#3  0x00005555555fb23d in std::function<bool (verible::VectorTree<verible::UnwrappedLine> const&)>::operator()(verible::VectorTree<verible::UnwrappedLine> const&) const (this=0x7fffffffcb00, __args#0=...)
    at /usr/include/c++/9/bits/std_function.h:688
#4  0x00005555555e3d5a in verible::FilterAlignablePartitions(verible::container_iterator_range<__gnu_cxx::__normal_iterator<verible::VectorTree<verible::UnwrappedLine>*, std::vector<verible::VectorTree<verible::UnwrappedLine>, std::allocator<verible::VectorTree<verible::UnwrappedLine> > > > > const&, std::function<bool (verible::VectorTree<verible::UnwrappedLine> const&)> const&) (range=..., ignore_partition_predicate=...)
    at common/formatting/align.cc:1065
#5  0x00005555555a4fa4 in verilog::formatter::ExtractAlignablePartitionGroups(const std::function<std::vector<verible::TaggedTokenPartitionRange, std::allocator<verible::TaggedTokenPartitionRange> >(const verible::container_iterator_range<__gnu_cxx::__normal_iterator<verible::VectorTree<verible::UnwrappedLine>*, std::vector<verible::VectorTree<verible::UnwrappedLine>, std::allocator<verible::VectorTree<verible::UnwrappedLine> > > > >&)> &, const verible::IgnoreAlignmentRowPredicate &, const verible::TokenPartitionRange &, const verilog::formatter::FormatStyle &) (group_extractor=..., ignore_group_predicate=..., full_range=..., vstyle=...)
    at verilog/formatting/align.cc:1294
#6  0x00005555555a5330 in verilog::formatter::AlignActualNamedParameters (full_range=..., vstyle=...) at verilog/formatting/align.cc:1325
#7  0x00005555555b536d in std::_Function_handler<std::vector<verible::AlignablePartitionGroup, std::allocator<verible::AlignablePartitionGroup> > (verible::container_iterator_range<__gnu_cxx::__normal_iterator<verible::VectorTree<verible::UnwrappedLine>*, std::vector<verible::VectorTree<verible::UnwrappedLine>, std::allocator<verible::VectorTree<verible::UnwrappedLine> > > > > const&, verilog::formatter::FormatStyle const&), std::vector<verible::AlignablePartitionGroup, std::allocator<verible::AlignablePartitionGroup> > (*)(verible::container_iterator_range<__gnu_cxx::__normal_iterator<verible::VectorTree<verible::UnwrappedLine>*, std::vector<verible::VectorTree<verible::UnwrappedLine>, std::allocator<verible::VectorTree<verible::UnwrappedLine> > > > > const&, verilog::formatter::FormatStyle const&)>::_M_invoke(std::_Any_data const&, verible::container_iterator_range<__gnu_cxx::__normal_iterator<verible::VectorTree<verible::UnwrappedLine>*, std::vector<verible::VectorTree<verible::UnwrappedLine>, std::allocator<verible::VectorTree<verible::UnwrappedLine> > > > > const&, verilog::formatter::FormatStyle const&) (
    __functor=..., __args#0=..., __args#1=...) at /usr/include/c++/9/bits/std_function.h:286
#8  0x00005555555b0ada in std::function<std::vector<verible::AlignablePartitionGroup, std::allocator<verible::AlignablePartitionGroup> > (verible::container_iterator_range<__gnu_cxx::__normal_iterator<verible::VectorTree<verible::UnwrappedLine>*, std::vector<verible::VectorTree<verible::UnwrappedLine>, std::allocator<verible::VectorTree<verible::UnwrappedLine> > > > > const&, verilog::formatter::FormatStyle const&)>::operator()(verible::container_iterator_range<__gnu_cxx::__normal_iterator<verible::VectorTree<verible::UnwrappedLine>*, std::vector<verible::VectorTree<verible::UnwrappedLine>, std::allocator<verible::VectorTree<verible::UnwrappedLine> > > > > const&, verilog::formatter::FormatStyle const&) const (this=0x555555977eb8, __args#0=..., __args#1=...) at /usr/include/c++/9/bits/std_function.h:688
#9  0x00005555555a64c5 in verilog::formatter::TabularAlignTokenPartitions (partition_ptr=0x5555559784e8, ftokens=0x7fffffffd2e8, full_text=..., disabled_byte_ranges=..., style=...) at verilog/formatting/align.cc:1437
#10 0x00005555555956c3 in verilog::formatter::Formatter::<lambda(verible::TokenPartitionTree&)>::operator()(verible::TokenPartitionTree &) const (__closure=0x5555559652f0, node=...) at verilog/formatting/formatter.cc:751
#11 0x0000555555596577 in std::_Function_handler<void(verible::VectorTree<verible::UnwrappedLine>&), verilog::formatter::Formatter::Format(const verilog::formatter::ExecutionControl&)::<lambda(verible::TokenPartitionTree&)> >::_M_invoke(const std::_Any_data &, verible::VectorTree<verible::UnwrappedLine> &) (__functor=..., __args#0=...) at /usr/include/c++/9/bits/std_function.h:300
#12 0x000055555559a77f in std::function<void (verible::VectorTree<verible::UnwrappedLine>&)>::operator()(verible::VectorTree<verible::UnwrappedLine>&) const (this=0x7fffffffd420, __args#0=...)
    at /usr/include/c++/9/bits/std_function.h:688
#13 0x0000555555598ad7 in verible::VectorTree<verible::UnwrappedLine>::ApplyPreOrder(std::function<void (verible::VectorTree<verible::UnwrappedLine>&)> const&) (this=0x5555559784e8, f=...) at ./common/util/vector_tree.h [... cut ...]
#18 0x0000555555595a35 in verilog::formatter::Formatter::Format (this=0x7fffffffd680, control=...) at verilog/formatting/formatter.cc:738
#19 0x00005555555939aa in verilog::formatter::FormatVerilog (text=..., filename=..., style=..., formatted_stream=..., lines=..., control=...) at verilog/formatting/formatter.cc:223
#20 0x00005555555737e7 in formatOneFile (filename=..., lines_to_format=...) at verilog/tools/formatter/verilog_format.cc:269
#21 0x0000555555574075 in main (argc=2, argv=0x7fffffffde48) at verilog/tools/formatter/verilog_format.cc:356
```

commit 8ee0a5c6 (2021-10-19): Fix crash caused by EOL comment between an entry and ','/';'

Fixes #920