chipsalliance/verible #950: preprocessor placement gives error

jefflfree, 2021-09-24:
**Describe the bug**

If I put `else in the middle of a module declaration (before ;) it fails. This is valid code.

**To Reproduce**
`ifndef three
module test1
`else
module test2
`endif
;
endmodule

Did it reject valid code? or crash?
./verible-verilog-lint blah.v
blah.v:3:1: syntax error, rejected "`else" (syntax-error).
blah.v:7:1: syntax error, rejected "endmodule" (syntax-error).

**Expected behavior**

A clear and concise description of what you expected to happen. Citations to the
[SystemVerilog-2017 Standard (LRM)](https://ieeexplore.ieee.org/document/8299595)
would help.

tgorochowik, 2021-09-24:
Hi!

Thanks for filing this issue. Unfortunately Verible does not handle preprocessing very well at this point. Also see: #183 

Having a proper preprocessor is certainly on our TODO list though

jefflfree, 2021-09-25:
I wonder if the preprocessor could be handled in the lexer instead of
tokenizing it in the parser?

Jeff

closed 2021-09-25 by jefflfree